steve-community/steve #1193: Disconected by powerloss,but state is still charging.

mtomoda, 2023-07-22:
#### Checklist

- [ X] I checked other issues already, but found no answer/solution
- [ X] I checked the documentation and wiki, but found no answer/solution
- [ X] I am running the latest version and the issue still occurs
- [ X] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.6.0
Operating system  : Debian
JDK               : ...
Database          : MariaDB
```

#### Expected Behavior
ChargePoint is not charging when power is restored after charging is interrupted by a power loss, but the state is still charging.

#### Actual Behavior
Since the connectors are connected, should the status be Preparing?

#### Steps to Reproduce the Problem
1.Start Charging.
2.Cut off power charge point.
3.Connect power charge point.

  1. ...
  2. ...
  3. ...

#### Additional context

...

mtomoda, 2023-07-22:
sorry wait a few minuits status changed available

[image]

juherr, 2023-07-24:
Could you close the issue if it is fixed?

goekay, 2023-07-25:
the information you see within steve comes from charging station. if _the latest communicated status_ is `Charging`, you will see `Charging`. even if the world burns down.

if the station communicates a status that represents the actual or more recent status, that one will be showed then.

closed 2023-08-20 by goekay