steve-community/steve #1204: SteVe does not work in combination with MariaDB 11

Smoerrebroed, 2023-08-01:
#### Checklist

- [X] I checked other issues already, but found no answer/solution
- [X] I checked the documentation and wiki, but found no answer/solution
- [X] I am running the latest version and the issue still occurs
- [X] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.6.0
Operating system  : [Docker]
JDK               : 20.0.1
Database          : MariaDB 11.0.2
```

#### Expected Behavior

SteVe starts up and successfully connects to its repository running on MariaDB 11.x

#### Actual Behavior

[ERROR] 2023-08-01 15:21:37,594 com.zaxxer.hikari.pool.HikariPool (main) - HikariPool-1 - Exception during pool initialization.
java.sql.SQLException: Unknown system variable 'transaction_isolation'
	at com.mysql.cj.jdbc.exceptions.SQLError.createSQLException(SQLError.java:129) ~[mysql-connector-java-8.0.30.jar

#### Steps to Reproduce the Problem

1. Configure and build SteVe
2. Provide MariaDB 11.x RDBMS as a backend
3. Look at log file

#### Additional context

The error can reproducibly be resolved by running on MariaDB 10.x

Smoerrebroed, 2023-08-07:
Is there anything else I need to do to so this gets looked at? IMHO it should be an easy fix: Updating the shipped JDBC driver to a more recent version - and potentially switching to MariaDB driver, particularly since that's the recommended database anyway.

goekay, 2023-08-12:
can confirm and reproduce.

> Updating the shipped JDBC driver to a more recent version

tired it. still error.

> potentially switching to MariaDB driver, particularly since that's the recommended database anyway.

this is something to think about. would not want to create compatibility issues with existing mysql installations. maybe we could start some branching: supporting `jdbc:mariadb://` alongside of `jdbc:mysql://...`. if mariadb is chosen, respective libs and config apply. otherwise (i.e. mysql) business as usual.

Smoerrebroed, 2023-08-18:
Here's a workaround: Configure your MariaDB 11.x to return 5.7.19 as version: https://mariadb.com/kb/en/server-system-variables/#version
Works for me!

juherr, 2023-08-18:
@goekay What is the plan? Do you think about 2 variants managed by 2 maven profiles or something else?

Smoerrebroed, 2024-11-28:
This seems to be fixed in the most recent release (probably due to updates in dependencies addressing the newer versions of MariaDB).

closed 2024-11-28 by Smoerrebroed