steve-community/steve #1205: queston / feature request

thorsten-gehrig, 2023-08-01:
Hi
I've tried to read/understand the configuration - but do not find anything on "notification settings".
Is there a way to send notification emails when the loading is stopped/completed to email and message (e.g. via ntfy.sh) the users?
I´m looking for a solution to notify our colleagues to ensure the EV-charger gets freed up for the next one if the charging is finished... guessing this is a very common need but i doesn't find a (free) solution yet...

Best
Thorsten

fnkbsi, 2023-08-02:
Hi Thorsten,
at the web frontend you can navigate to the settings, where you can find a section "Mail Notification Setting".
There you can set the  eMail account parameters from an dedicated Mail-Server. Also you can define the recipient (I am not sure if you can use more than one, try a comma separated list). The last step is to choose the events, when a mail shall be send.

At the moment SteVe provides no option to send a mail only to the user of the ended transaction.

There is one more thing. Usually the transaction ends when the connecter is disconnected, not when the current is zero or the EV is back in State B. In that case most charge boxes sending a StatusNotification with SuspendedEV, unfortunately SteVe provides this actually not as Event.

Hope this helps.

goekay, 2023-08-12:
> Also you can define the recipient (I am not sure if you can use more than one, try a comma separated list)

https://github.com/steve-community/steve/blob/master/src/main/resources/webapp/WEB-INF/views/settings.jsp#L57 :wink:

closed 2023-08-29 by goekay

thorsten-gehrig, 2023-10-12:
Hi 
Thank you @fnkbsi and @goekay for your replies.
Seems I´ve misunderstood some details earlier... and i will probably build a test-environment soon.

As I've understood push-messages are not yet available - Is there a chance that we get a push-notification e.g. via ntfy.sh added to Steve? (it´s "just" a shell-command line to be added in parallel to the email configuration).

Furthermore I´ve understood SuspendedEV event isn't yet an event - adding SuspendedEV as an event would be great (so emails / push-notifications based on that can be send). 
Any chance for this added? Is there a specific reason "SuspendedEV" event isn't yet part of the events?

If there is a sponsorship required please let me know...

Best
Thorsten

fnkbsi, 2023-10-12:
Hi Thorsten, 
have a look on PK #1263. You will still miss ntsy.sh

thorsten-gehrig, 2023-10-12:
Great - thanks! ( I owe you a beer / coffee ....)
I´m not a java expert... but may i find the line of code where the emails are composed... 
Finally the ntfy.sh is as easy raisin a shell-command like
curl   -d "<event>"  ntfy.sh/mytopic

If you replace the "mytopic" by <email-adress> it should have the job done...
Obvisouly replacing <event> by a more readable Message to the user whould be more beautify.

As a side note: is there any chance to change the email-template/email-content which is send to the user?

Best
Thorsten

fnkbsi, 2023-10-12:
The emails are composed in service/NotificationService.java. You can adapt the methods createContent(...) in the same class if you want to change the mail content.

thorsten-gehrig, 2023-10-14:
Thanks. 
I'm able to do this... but still having some challenges.
a) Any timeline for the next release (with the merge from fnkbsi)?
b)I´m struggeling with the emails itself. 
There is an email-config at the settings - there I can provide multiple email address (one per line), correct?
I would assume this is the mailbox for the fleet manager (who operates Steve and the charging stations).
This works fine for me.
And I can configure an email-address within each user account - this seems not working consistent for me.
I would assume the email address of the user is used for notifications related to this user-id, isn't?
(bonus question: can I provide here also 2 addresses for 1 user? the nifty.sh does also have an email-trigger ;-) ).
The help isn't very detailed - and I don't found any documentation ... so any help is appreciated.

Best
Thorsten

fnkbsi, 2023-10-15:
"There is an email-config at the settings - there I can provide multiple email address (one per line), correct?" Yes
"I would assume this is the mailbox for the fleet manager (who operates Steve and the charging stations)." Managers if multiple addresses are configured
"And I can configure an email-address within each user account - this seems not working consistent for me." Any suggestions?
"I would assume the email address of the user is used for notifications related to this user-id, isn't?"
That's the case in PR #1263, but a user only get's EVSuspend and TransactionStop notifications. A user can have multiple email addresses, comma separated.

thorsten-gehrig, 2023-10-15:
Thank you - appreciated!

thorsten-gehrig, 2023-10-15:
Multiple addresses doesn't work - I get an error message "Not a valid e-mail address" - seems there is a validation check not expecting coma separated email-addresses...

fnkbsi, 2023-10-16:
> Multiple addresses doesn't work - I get an error message "Not a valid e-mail address" - seems there is a validation check not expecting coma separated email-addresses...
On adding the addresses to the user, right? Sorry, missed that.

thorsten-gehrig, 2023-10-16:
Is there another way to add multiple email? Or will you update this?
Thank you in advance.

fnkbsi, 2023-10-17:
If your mail-server supports distribution-lists (a mail-address with multiple recipients), you can use them.
I don't have any updates planned at this time

thorsten-gehrig, 2023-10-22:
Thanks - but this doesn't help me and DLs are not an option.
If I understand correctly it's just the validation of the input - I might be able to identify the area and disable the validation check myself...

fnkbsi, 2023-10-23:
Can't recommend it! 

Have a look at the annotations here
https://github.com/steve-community/steve/blob/master/src/main/java/de/rwth/idsg/steve/web/dto/UserForm.java