steve-community/steve #1228: Unable to Run Steve with Remote IP Address Binding

xiaojieTao, 2023-08-30:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [x] I am running the latest version and the issue still occurs
- [x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.6.0
Operating system  : Ubuntu 20.04 LTS VM on Windows 10 host.
JDK               : JDK 11
Database          : MariaDB 10.4
```

#### Expected Behavior
Steve should build and run successfully when configured to bind to the actual IP address of the machine. 
...

#### Actual Behavior
When configuring Steve to bind to the loopback address (127.0.0.1), the server successfully builds and runs. However, when attempting to configure the server to bind to the actual IP address of the machine (to listen to remote devices), the server builds successfully but does not run as expected.
...

#### Steps to Reproduce the Problem

  1. Database preparation, download and extract tarball
  2. Modify the [configuration ](https://github.com/steve-community/steve/blob/master/src/main/resources/config/prod/main.properties#L28)to bind the server to the actual IP address of the machine.
  3. Build Steve.
  4. Attempt to run Steve.
  5. Observe that the server fails to start.

#### Additional context
The Ubuntu console messages right after attempting to run Steve
```
xiaojietao@xiaojietao-VirtualBox:/steve-steve-3.6.0$ java -jar target/steve.jar
Log file: /home/xiaojietao/logs/steve.log
Starting. FAILED!
Please refer to the log file for details
Exception in thread "main" java.util.concurrent.ExecutionException: java.lang.IllegalStateException: STOPPED
	at java.base/java.util.concurrent.CompletableFuture.reportGet(CompletableFuture.java:395)
	at java.base/java.util.concurrent.CompletableFuture.get(CompletableFuture.java:2022)
	at org.eclipse.jetty.server.Server.doStop(Server.java:490)
	at org.eclipse.jetty.util.component.AbstractLifeCycle.stop(AbstractLifeCycle.java:132)
	at de.rwth.idsg.steve.JettyServer.stop(JettyServer.java:185)
	at de.rwth.idsg.steve.SteveProdStarter.stop(SteveProdStarter.java:80)
	at de.rwth.idsg.steve.Application.stop(Application.java:85)
	at de.rwth.idsg.steve.Application.main(Application.java:68)
Caused by: java.lang.IllegalStateException: STOPPED
	at org.eclipse.jetty.server.handler.ContextHandler.shutdown(ContextHandler.java:774)
	at java.base/java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:195)
	at java.base/java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1655)
	at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:484)
	at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:474)
	at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:550)
	at java.base/java.util.stream.AbstractPipeline.evaluateToArrayNode(AbstractPipeline.java:260)
	at java.base/java.util.stream.ReferencePipeline.toArray(ReferencePipeline.java:517)
	at org.eclipse.jetty.util.component.Graceful.shutdown(Graceful.java:146)
	... 6 more

```

The log right after attempting to run Steve and failed
```

[INFO ] 2023-08-30 00:13:22,260 de.rwth.idsg.steve.utils.PropertiesFileLoader (main) - Hint: The Java system property 'main.properties' can be set to point to an external properties file, which will be prioritized over the bundled one
[INFO ] 2023-08-30 00:13:22,398 de.rwth.idsg.steve.Application (main) - Loaded the properties. Starting with the 'PROD' profile
[INFO ] 2023-08-30 00:13:22,448 de.rwth.idsg.steve.Application (main) - Date/time zone of the application is set to UTC. Current date/time: 2023-08-30T00:13:22.434Z
[INFO ] 2023-08-30 00:13:22,790 org.eclipse.jetty.server.Server (main) - jetty-10.0.14; built: 2023-02-22T23:12:32.272Z; git: 976721d0f3e903a243584d47870ad2f2c1bf9e55; jvm 11.0.20+8-post-Ubuntu-1ubuntu120.04

```
...

pjjjv, 2023-09-06:
I see this too.

xiaojieTao, 2023-09-06:
> e

Have you fixed the problem? I'm still stuck with it.

xiaojieTao, 2023-09-20:
Please heeeeeeeeelp!

juherr, 2023-09-21:
Could you try again with the latest commit on master?
You should have more details to share in logs.

pjjjv, 2023-09-21:
Could you try to put back:
server.host = 0.0.0.0
?

xiaojieTao, 2023-09-29:
@pjjjv @juherr 

Thank you both for the help. I can now run Steve in Docker on a MacBook M1 with the latest commit but I still have a question regarding IP address and URL.
I docker-compose with the default configuration and the the URL is 
```
2023-09-26 15:37:18 steve-app-1  |  Done!
2023-09-26 15:37:18 steve-app-1  | Hint: You can stop the application by pressing CTRL+C
2023-09-26 15:37:18 steve-app-1  | 
2023-09-26 15:37:18 steve-app-1  | Access the web interface using
2023-09-26 15:37:18 steve-app-1  | - http://172.24.0.3:8180/steve/manager/home
2023-09-26 15:37:18 steve-app-1  | SOAP endpoint for OCPP
2023-09-26 15:37:18 steve-app-1  | - http://172.24.0.3:8180/steve/services/CentralSystemService
2023-09-26 15:37:18 steve-app-1  | WebSocket/JSON endpoint for OCPP
2023-09-26 15:37:18 steve-app-1  | - ws://172.24.0.3:8180/steve/websocket/CentralSystemService/(chargeBoxId)
```
However, the URL 'http://172.24.0.3:8180/steve/manager/home' doesn't open. I have to put my Mac's Local IP address in the URL 'http://192.168.10.170:8180/steve/manager/home' to access Steve's home page. Is this the expected behavior of Steve? How can I access Steve now through internet instead of the LAN of one router only?

I know very little about computer network and I apologize for any misuse of its terminology.

slachiewicz, 2023-09-29:
This is how it works inside containers, not too much You can do. 

If You need to run an instance available on the internet - run Steve on Ckoud, sample instructions here: https://github.com/steve-community/steve/wiki/Create-SteVe-Instance-in-AWS-Lightsail 

or solutions like ngrok https://medium.com/weekly-webtips/lets-expose-a-local-webserver-to-the-internet-with-ngrok-c14646fb7821

xiaojieTao, 2023-10-07:
I tried setting up port forwarding on my router (8180 forwarded to 8180) but I still cannot access the Steve admin page through my router's public IP address (http://<my router's public IP>:8180/steve/manager/signin). Is this a bug of Steve or I am doing something wrong?
There is no one talking about port forwarding on this repo so I feel I shouldn't use port forwarding to expose Steve to public internet.

goekay, 2026-02-24:
closing due to inactivity.