steve-community/steve #1232: SteVe stops working

elektroscout, 2023-09-05:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [x] I am running the latest version and the issue still occurs
- [x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : .3.6.0
Operating system  : Ubuntu 20.04.6
JDK               : ...
Database          : 1.0.3
```

#### Expected Behavior

After starting steve.jar SteVe keeps running

#### Actual Behavior

After starting steve.jar it works fine but after a shlort time SteVe stops running. I can start it again but it keeps stopping after time.

#### Steps to Reproduce the Problem

  1. Start SteVe: -jar target/steve.jar
  2. wait till it stops.
  3. Start SteVe again

#### Additional context

The system was setup using the guide for Ubuntu. Everything works fine including adding CP and managing them. The only thing is: SteVe stops running after time. I did not find anything about this issue in the open issues. I don't think this is something big but I could not find out what to do to solve this. Can you help me?

Thanks in advance &
Regards
André

juherr, 2023-09-05:
Could you share the logs before the crash?

goekay, 2024-02-18:
closing due to inactivity, and no context/logs provided in order to investigate.

closed 2024-02-18 by goekay

elektroscout, 2024-05-11:
Hi Sevket

After a while we decided to pick up this project again.
Is there someone you know that can help us to install the server properly? With the manual I manage to setup the server but there are two issues we need help (of course we will pay for professional help)

  *
setup https / wss
  *
keep the application running after closing the terminal (putty)

These are maybe simple issues for a linux expert, but we are not (-:

Can you help or refer to someone who can?

I would appreciate your feedback.

Regards
André

goekay, 2024-05-11:
Pls drop me an email, and we can arrange a meeting for setup.

elektroscout, 2024-05-11:
Hi Sevket

Thanks for your prompt reply, cool!

When is the most convenient time/date for you? I adjust to your availability.

Regards
André