steve-community/steve #1236: Error while trying to add a charging profile: Schedule Periods cannot be empty

FredBoard, 2023-09-10:
#### Checklist

- [ x] I checked other issues already, but found no answer/solution
- [ x] I checked the documentation and wiki, but found no answer/solution
- [ x] I am running the latest version and the issue still occurs
- [ x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.6.0
Operating system  : 1.0.3 ( 2023-09-07 at 07:13 )
JDK               : Linux OCPP 5.10.43v64 #1 SMP PREEMPT Tue Jun 15 00:52:48 CEST 2021 aarch64 GNU/Linux ( Raspberry )
Database       : mariadb  Ver 15.1 Distrib 10.3.39-MariaDB, for debian-linux-gnu (aarch64) using readline 5.2
```

#### Expected Behavior

No timeout

#### Actual Behavior

Im Client (EVlink Schneider Electric)  logfile,  i found : 

Sun Sep 10 09:45:44 2023 | Sending StartTransactionReq tag:045924DA6C7480
Sun Sep 10 09:46:44 2023 | StartTransactionReq: error (InternalError, Timeout waiting reply, )

#### Steps to Reproduce the Problem

  1. Click "DATA MANAGMENT"
  2. Select "CHARGING PROFILES"
  3. Click "ADD NEW"
  4. Give all mandatory fileds. ( Form with 8 fields )
  5. Clik ADD

=> Schedule Periods cannot be empty

I do not have found documentation about what is exactly a "Charging Profile" ....
  

#### Additional context

...

FredBoard, 2023-09-11:
I do not know why the apps add a zero before rfid tag :

`SELECT * FROM ocpp_tag WHERE id_tag like '%45924DA6C7480'; `
ocpp_tag_pk, id_tag, parent_id_tag, expiry_date, max_active_transaction_count, note
```
(1,   '45924DA6C7480',  NULL, NULL, 1, 'Test Badge'),
(17, '045924DA6C7480', NULL, NULL, 0, 'This unknown idTag was used in a transaction that started @ 2023-09-11T09:44:21.000Z. It was reported @ 2023-09-11T09:44:22.140Z.');
```
result 

```
[INFO ] 2023-09-09 22:18:19,695 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger (qtp695530017-150) - [chargeBoxId=3N_22031_25_001_011, sessionId=22e70877-4c78-77aa-4b70-1bb4b6ed0304] Received: [2,"20","Authorize",{"idTag":"045924DA6C7480"}]
[ERROR] 2023-09-09 22:18:19,700 de.rwth.idsg.steve.service.OcppTagService (qtp695530017-150) - The user with idTag '045924DA6C7480' is INVALID (not present in DB).
```

juherr, 2023-09-11:
@FredBoard 

Did you fix your timeout issue? If yes, what was the problem and its fix?

> I do not know why the apps add a zero before rfid tag :

It is not the app but the station. 
I'm surprised your tag has an odd length and I bet your initial value comes from a tool that left trim `0`.
You should be able to validate the tag with any rfid reader.

goekay, 2024-05-20:
this issue mixes two topics: 

1. wrong usage/reporting of ocpp id tag by the station. as a consequence, steve does the right thing marks it as invalid. nothing to do with steve.
2. charging profile and scheduled period cannot be empty. don't know what this has to do with (1).

nevertheless, closing due to inactivity.

closed 2024-05-20 by goekay