steve-community/steve #1239: Build failure when compiling SteVe

remyzerems, 2023-09-12:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [x] I am running the latest version and the issue still occurs
- [x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.6.0 (3.4.5 also impacted)
Operating system  : Ubuntu 22.04
JDK               : 11
Database          : MariaDB 10.4
```

#### Expected Behavior

Running `docker-compose up` against the official release should build SteVe successfully

#### Actual Behavior

An error occurs while building. It crashes on the Flyway Migration while trying to migrate from v0.7.6 to v0.7.7.

```
app_1  | [INFO] Flyway Community Edition 7.15.0 by Redgate
app_1  | [INFO] Database: jdbc:mysql://mariadb:3306/stevedb (MySQL 5.5)
app_1  | [INFO] Successfully validated 35 migrations (execution time 00:00.043s)
app_1  | [INFO] Creating Schema History table `stevedb`.`schema_version` ...
app_1  | [INFO] Current version of schema `stevedb`: << Empty Schema >>
app_1  | [WARNING] outOfOrder mode is active. Migration of schema `stevedb` may not be reproducible.
app_1  | [INFO] Migrating schema `stevedb` to version "0.6.6 - inital"
app_1  | [WARNING] DB: Event scheduler is switched off, use SET GLOBAL event_scheduler=ON to enable it. (SQL State: HY000 - Error Code: 1105)
app_1  | [INFO] Migrating schema `stevedb` to version "0.6.7 - update"
app_1  | [WARNING] DB: Event scheduler is switched off, use SET GLOBAL event_scheduler=ON to enable it. (SQL State: HY000 - Error Code: 1105)
app_1  | [INFO] Migrating schema `stevedb` to version "0.6.8 - update"
app_1  | [INFO] Migrating schema `stevedb` to version "0.6.9 - update"
app_1  | [INFO] Migrating schema `stevedb` to version "0.7.0 - update"
app_1  | [INFO] Migrating schema `stevedb` to version "0.7.1 - update"
app_1  | [INFO] Migrating schema `stevedb` to version "0.7.2 - update"
app_1  | [INFO] Migrating schema `stevedb` to version "0.7.3 - update"
app_1  | [INFO] Migrating schema `stevedb` to version "0.7.6 - update"
app_1  | [INFO] Migrating schema `stevedb` to version "0.7.7 - update"
app_1  | [ERROR] Migration of schema `stevedb` to version "0.7.7 - update" failed! Please restore backups and roll back database and code!
app_1  | [INFO] ------------------------------------------------------------------------
app_1  | [INFO] BUILD FAILURE
app_1  | [INFO] ------------------------------------------------------------------------
app_1  | [INFO] Total time:  30.723 s
app_1  | [INFO] Finished at: 2023-09-12T07:49:20Z
app_1  | [INFO] ------------------------------------------------------------------------
app_1  | [WARNING] 
app_1  | [WARNING] Plugin validation issues were detected in 2 plugin(s)
app_1  | [WARNING] 
app_1  | [WARNING]  * pl.project13.maven:git-commit-id-plugin:4.9.10
app_1  | [WARNING]  * org.flywaydb:flyway-maven-plugin:7.15.0
app_1  | [WARNING] 
app_1  | [WARNING] For more or less details, use 'maven.plugin.validation' property with one of the values (case insensitive): [BRIEF, DEFAULT, VERBOSE]
app_1  | [WARNING] 
app_1  | [ERROR] Failed to execute goal org.flywaydb:flyway-maven-plugin:7.15.0:migrate (default) on project steve: org.flywaydb.core.internal.command.DbMigrate$FlywayMigrateException: Migration V0_7_7__update.sql failed
app_1  | [ERROR] -----------------------------------
app_1  | [ERROR] SQL State  : HY000
app_1  | [ERROR] Error Code : 1833
app_1  | [ERROR] Message    : Cannot change column 'chargeBoxId': used in a foreign key constraint 'stevedb/FK_chargeBoxId_c' of table 'stevedb/connector'
app_1  | [ERROR] Location   : /code/src/main/resources/db/migration/V0_7_7__update.sql (/code/src/main/resources/db/migration/V0_7_7__update.sql)
app_1  | [ERROR] Line       : 1
app_1  | [ERROR] Statement  : ALTER TABLE `chargebox`
app_1  | [ERROR] CHANGE COLUMN `chargeBoxId` `chargeBoxId` VARCHAR(255) NOT NULL COMMENT '' ,
app_1  | [ERROR] CHANGE COLUMN `endpoint_address` `endpoint_address` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
app_1  | [ERROR] CHANGE COLUMN `ocppProtocol` `ocppProtocol` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
app_1  | [ERROR] CHANGE COLUMN `chargePointVendor` `chargePointVendor` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
app_1  | [ERROR] CHANGE COLUMN `chargePointModel` `chargePointModel` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
app_1  | [ERROR] CHANGE COLUMN `chargePointSerialNumber` `chargePointSerialNumber` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
app_1  | [ERROR] CHANGE COLUMN `chargeBoxSerialNumber` `chargeBoxSerialNumber` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
app_1  | [ERROR] CHANGE COLUMN `fwVersion` `fwVersion` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
app_1  | [ERROR] CHANGE COLUMN `fwUpdateStatus` `fwUpdateStatus` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
app_1  | [ERROR] CHANGE COLUMN `iccid` `iccid` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
app_1  | [ERROR] CHANGE COLUMN `imsi` `imsi` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
app_1  | [ERROR] CHANGE COLUMN `meterType` `meterType` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
app_1  | [ERROR] CHANGE COLUMN `meterSerialNumber` `meterSerialNumber` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
app_1  | [ERROR] CHANGE COLUMN `diagnosticsStatus` `diagnosticsStatus` VARCHAR(255) NULL DEFAULT NULL COMMENT ''
app_1  | [ERROR] 
app_1  | [ERROR] -> [Help 1]
app_1  | [ERROR] 
app_1  | [ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
app_1  | [ERROR] Re-run Maven using the -X switch to enable full debug logging.
app_1  | [ERROR] 
app_1  | [ERROR] For more information about the errors and possible solutions, please read the following articles:
app_1  | [ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException
test_steve_install_app_1 exited with code 1
```

#### Steps to Reproduce the Problem

  1. Download and extract 3.6.0 release from [https://github.com/steve-community/steve/archive/refs/tags/steve-3.6.0.tar.gz](https://github.com/steve-community/steve/archive/refs/tags/steve-3.6.0.tar.gz)
  2. Run `docker-compose up` from the folder containing the docker-compose.yml

slachiewicz, 2023-09-12:
See #1233 for suggestions.

To fix it we must do a new release.

remyzerems, 2023-09-12:
Many thanks for your quick answer !

I changed MariaDB to 10.4.30 as stated in the referenced issue and it worked treat !

Simply added ".30" to docker-compose.yml for mariadb: "image: mariadb:10.4<b>.30</b>"

closed 2023-09-12 by remyzerems