steve-community/steve #1245: Allow All OCPP TAGS

jot4p, 2023-09-20:
Hi,
I have some EV chargers and multiple tags.

It's possible to allow always any type of tag ? Any user with compatible RF card (tag) can charge the vehicle!

juherr, 2023-09-20:
Why do you use tag if every one is free to charge? 
Why not configuring your charger into a plug-n-charge mode where it will send a default tag value configured in steve?

jot4p, 2023-09-20:
Hi Juherr ,
How I can adjust my charger to plug and charge mode ? That is the question :)

juherr, 2023-09-21:
You should provide the manifacturer, model and firmware version of your charger.

But nobody can replace the official support and you should start with them.

jot4p, 2023-09-21:
Hi juherr

I thought it was a parameter that I could send via ocpp /STEVE

[image]

goekay, 2023-09-21:
i dont think there is an out of the box feature in ocpp standard that will provide this. AFAIK though. i might have a blind spot.

what you can do is (with implementation effort): always react to authorize requests with "Accepted" for every incoming id tag.

juherr, 2023-09-21:
> I thought it was a parameter that I could send via ocpp /STEVE

Maybe an ocpp key exists to configure the charger but again you should ask to the charger manifacturer.

From [the datasheet](https://circontrol.com/datasheets/evolvesmart-eng.pdf):
[image]

What you are asking for sounds like the "Connect 'n' Charge" mode.

jot4p, 2023-10-03:
Ok, 
Instead of use Open Access , maybe I need to develop a APP to request charging activation..  What is the best way to create the app, based on Steve Websocket ?

juherr, 2023-10-03:
You'll have to find where is the authorization logic and replace it by your own.

fnkbsi, 2023-10-11:
An alternativ to @juherr suggestion is to add the OCPP operations "Remote Start Transaction" and "Remote Stop Transaction" to the WebApi. Then you can use the steve web interface for your APP.

juherr, 2023-10-11:
Yes, see #1000

goekay, 2024-08-07:
i consider the issue resolved, since multiple solution approaches have been suggested even though steve does not support this out of the box. this is a too nuanced feature to implement globally in steve. there are many use cases. @jot4p wants all tags to be allowed at all stations. what if some other user wants all tags to be allowed at _some_ stations? etc.

closed 2024-08-07 by goekay