steve-community/steve #1256: Cannot Unlock Charger When It's Connected to Steve

xiaojieTao, 2023-10-03:
I connected my Wallbox Pulsar Plus charger to Steve OCPP and the charger turned itself to 'Locked' status and I can't figure out how to unlock it. I tried 'Unlock Connector' operation in Steve and the response from charger is 'NotSupported'. I think this operation is to unlock any physical connector retention lock and my charger doesn't have one. I browsed the Wallbox documentations and I found these words in [this page](https://support.wallbox.com/en/knowledge-base/ocpp-activation-and-setup-guide/):

_Unlocking a charger
When your charger is connected via OCPP to your CSMS, users will be able to unlock the charger by:
Using an RFID card
Introducing a PIN code (Commander 2 only)_

However, my model (Pulsar Plus) doesn't have a RFID reader and therefore neither of the option works. How can I unlock the charger?

juherr, 2023-10-03:
That's a question for the support of pulsar.

Maybe try to do it via the mobile app.
Or https://support.wallbox.com/en/knowledge-base/how-to-restore-your-charger/

xiaojieTao, 2023-10-03:
> That's a question for the support of pulsar.
> 
> Maybe try to do it via the mobile app. Or https://support.wallbox.com/en/knowledge-base/how-to-restore-your-charger/

I contacted Wallbox customer service and they told me that once the charger is connected to OCPP, the OCPP backend is responsible for unlocking the charger. Is there any feature or operation in Steve related to unlocking a charger?

goRaspy, 2023-10-04:
Try change availability ocpp command

juherr, 2023-10-04:
> I contacted Wallbox customer service and they told me that once the charger is connected to OCPP, the OCPP backend is responsible for unlocking the charger. Is there any feature or operation in Steve related to unlocking a charger?

Ask them what is the expected ocpp request and we will be able to say how to produce it.

goekay, 2023-10-04:
i think the wording "unlock" might be unfortunate here which leads you to a certain path.

@xiaojieTao what you want is probably _Remote Start Transaction_: you want to remotely authorize a user to charge since there is no other option (i.e. rfid card).

xiaojieTao, 2023-10-07:
Fix: It turned out that Remote Start Transaction is the expected OCPP operation and it can 'unlock' the charger.
Appreciate the comments above!!

closed 2023-10-15 by goekay