steve-community/steve #1259: BootNotification validation does not enforce maxLength for chargePointModel

Tano-Coppoletta, 2023-10-08:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [] I am running the latest version and the issue still occurs
- [x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.6.0
Operating system  : Ubuntu 22.04.1
JDK               : openjdk version "11.0.20.1"
Database          : version 1.0.3
```

#### Expected Behavior

When sending a "BootNotification" message that does not comply with the json schema, the server should send a response with code 4 and an error.

#### Actual Behavior
The server responded with a valid response (status: Accepted)

#### Steps to Reproduce the Problem

  1. Craft a "BootNotification" message containing the property chargePointModel with a lenght greater than 20
  2. Send it to Steve
  3. Look at the response

ahzf, 2023-10-08:
That is really not a bug... except perhaps within OCPP as this artificial limitation really makes no sense in 2023++ ;)

goekay, 2023-11-07:
steve enforces a length of 255 for fields from BootNotification. we started following the spec verbatim, but had to relax these  constraints since station vendors were not respecting the limits.

goekay, 2023-12-03:
i will close this ticket as the missing compliance is explained in the message above. any security-related consequences or vulnerabilities are not mentioned. therefore, i conclude that this is a benign situation.

closed 2023-12-03 by goekay