steve-community/steve #1261: Can't build SteVe

warukunaiyone, 2023-10-10:
I have followed the steps. But stuck on Build (SteVe 3.6.0)
System Requirements (Installed)

- JDK 11
- Maven 3.9.4
- MariaDB 10.5

I have created a database, changed the Timezone and made changes in main.properties
when cd to steve and run mvn package

[steve err.txt](https://github.com/steve-community/steve/files/12854750/steve.err.txt)

and when run mvn flyway:repair

[image]

Am I missing something?

juherr, 2023-10-10:
What is the command line you ran?

warukunaiyone, 2023-10-11:
> What is the command line you ran?
I just edited a post. Please check.

juherr, 2023-10-11:
You miss to pass the database properties: `-Ddb.xxx=yyy`

warukunaiyone, 2023-10-11:
> You miss to pass the database properties: `-Ddb.xxx=yyy`

Where should I fix it?

warukunaiyone, 2023-10-11:
> you should figure out why `mvn package` fails in the first place. i have a feeling that it is because of mariadb 10.5.X you are using. mariadb introduced some breaking changes in recent versions. i dont know which patch version X you are using but i would suggest to stick to one of the tested-and-working versions: `'mariadb:10.4.30', 'mariadb:10.5.21', 'mariadb:10.6.14'`
> 
> that aside, `mvn flyway:repair` fails because properties from the properties file of the env are not available. in order to make them available you need to add initialize as first command, as in `mvn initialize flyway:repair`. this will fix the property problem, but you will probably still fail to because of the first paragraph. try to pin the mariadb version.

Thanks for the answer, I'll try it.

warukunaiyone, 2023-10-11:
> you should figure out why `mvn package` fails in the first place. i have a feeling that it is because of mariadb 10.5.X you are using. mariadb introduced some breaking changes in recent versions. i [...]

When I switched to MariaDB version 10.4.30 I was able to build and run it. Thank you very much for your help.

closed 2023-10-11 by warukunaiyone