steve-community/steve #1262: Charger doesnt appear on website

plstah, 2023-10-11:
[image]

Hi folks. 
Please tell me if i have done correctly this settings because my charger doesnt appear on the Steve website.
[image]

Thank you for your help.

juherr, 2023-10-11:
What is the full url of the configured websocket?
Do you have anything in the "unknown station" hidden section?
What are the manufacturer, model and firmware version of the station?

plstah, 2023-10-12:
Hi there.
I have nothing in the section " Unknown Charge Points  ".
The url is : ws://192.168.226.5:3306/steve/websocket/CentralSystemService
Please tell me at the protocol section is correct what i wrote? Ocpp1.6
Find attached the model of the charger.

Regards.

warukunaiyone, 2023-10-12:
Hi,
I have the same problem, I think I entered the wrong AdminAddress.
Where can I reference the AdminAddress from the charger?
[image]

fnkbsi, 2023-10-12:
@plstah: I guess you are using the wrong port. Port 3306 belongs to the database not to the webserver.

On the Steve-Site under "About" of your server you can find the url of the websocket

plstah, 2023-10-12:
Thanks, I'll check it out.

warukunaiyone, 2023-10-13:
> /steve/websocket/CentralSystemService
Which format is correct? Thank you.
Admin Address : (ipcharger):8080/steve/websocket/CentralSystemService/ or (ipcharger):8080

plstah, 2023-10-13:
I have modified the port number, but nothing happens :(

What else should i do?

Thank you!

fnkbsi, 2023-10-13:
Difficult to say. I would start with checking that "Unkonwn charge point" list is not hidden (click on the text). Then check the log. 
After that check the url registered in the charger, some need a backslash at the end other don't.  And check everything again.
Also make sure no firewall blocks your charge from connecting steve.

fnkbsi, 2023-10-13:
> > /steve/websocket/CentralSystemService
> > Which format is correct? Thank you.
> > Admin Address : (ipcharger):8080/steve/websocket/CentralSystemService/ or (ipcharger):8080

the first format

plstah, 2023-10-13:
The protocol is correct? Please tell me this.

goekay, 2023-10-15:
admin address in the chargebox page is not relevant.

do you see any events/actions in the logs?

plstah, 2023-10-16:
I have checked all the things, but unfortunately the charger doesn`t appear.
Below i send you the log.

"[INFO ] 2023-10-13 07:53:58,463 org.eclipse.jetty.server.Server
(JettyShutdownThread) - Stopped ***@***.***{STOPPING}[10.0.14,sto=5000]

[INFO ] 2023-10-13 07:53:58,464 org.eclipse.jetty.server.Server
(JettyShutdownThread) - Shutdown
***@***.***{STOPPING}[10.0.14,sto=5000]
[INFO ] 2023-10-13 07:53:58,557
org.eclipse.jetty.server.AbstractConnector (JettyShutdownThread) -
Stopped ***@***.***{HTTP/1.1, (http/1.1)}{0.0.0.0:8080}
[INFO ] 2023-10-13 07:53:58,582
org.eclipse.jetty.server.handler.ContextHandler.steve
(JettyShutdownThread) - Destroying Spring FrameworkServlet
'spring-dispatcher'
[INFO ] 2023-10-13 07:53:58,728 com.zaxxer.hikari.HikariDataSource
(JettyShutdownThread) - HikariPool-1 - Shutdown initiated...
[INFO ] 2023-10-13 07:53:58,825 com.zaxxer.hikari.HikariDataSource
(JettyShutdownThread) - HikariPool-1 - Shutdown completed.
[INFO ] 2023-10-13 07:54:14,796
de.rwth.idsg.steve.utils.PropertiesFileLoader (main) - Hint: The Java
system property 'main.properties' can be set to point to an external
properties file, which will be prioritized over the bundled one
[INFO ] 2023-10-13 07:54:15,230 de.rwth.idsg.steve.Application (main)
- Loaded the properties. Starting with the 'PROD' profile
[INFO ] 2023-10-13 07:54:15,350 de.rwth.idsg.steve.Application (main)
- Date/time zone of the application is set to UTC. Current date/time:
2023-10-13T07:54:15.287Z
[INFO ] 2023-10-13 07:54:17,048 org.eclipse.jetty.server.Server (main)
- jetty-10.0.14; built: 2023-02-22T23:12:32.272Z; git:
976721d0f3e903a243584d47870ad2f2c1bf9e55; jvm
11.0.20+8-post-Raspbian-1deb11u1
[INFO ] 2023-10-13 07:54:18,216
org.eclipse.jetty.server.handler.ContextHandler.steve (main) - No
Spring WebApplicationInitializer types detected on classpath
[INFO ] 2023-10-13 07:54:18,428
org.eclipse.jetty.server.session.DefaultSessionIdManager (main) -
Session workerName=node0
[INFO ] 2023-10-13 07:54:18,489
org.eclipse.jetty.server.handler.ContextHandler.steve (main) -
Initializing Spring root WebApplicationContext
[INFO ] 2023-10-13 07:54:18,490
org.springframework.web.context.ContextLoader (main) - Root
WebApplicationContext: initialization started
[INFO ] 2023-10-13 07:54:23,211 com.zaxxer.hikari.HikariDataSource
(main) - HikariPool-1 - Starting...
[INFO ] 2023-10-13 07:54:24,125 com.zaxxer.hikari.pool.HikariPool
(main) - HikariPool-1 - Added connection
***@***.***
[INFO ] 2023-10-13 07:54:24,133 com.zaxxer.hikari.HikariDataSource
(main) - HikariPool-1 - Start completed.
[INFO ] 2023-10-13 07:54:32,794
org.apache.cxf.wsdl.service.factory.ReflectionServiceFactoryBean
(main) - Creating Service {urn://Ocpp/Cs/2010/08/}CentralSystemService
from class ocpp.cs._2010._08.CentralSystemService
[INFO ] 2023-10-13 07:54:49,095 org.apache.cxf.endpoint.ServerImpl
(main) - Setting the server's publish address to be
/CentralSystemServiceOCPP12
[INFO ] 2023-10-13 07:54:49,146
org.apache.cxf.wsdl.service.factory.ReflectionServiceFactoryBean
(main) - Creating Service {urn://Ocpp/Cs/2012/06/}CentralSystemService
from class ocpp.cs._2012._06.CentralSystemService
[INFO ] 2023-10-13 07:54:49,386 org.apache.cxf.endpoint.ServerImpl
(main) - Setting the server's publish address to be
/CentralSystemServiceOCPP15
[INFO ] 2023-10-13 07:54:49,389
org.apache.cxf.wsdl.service.factory.ReflectionServiceFactoryBean
(main) - Creating Service {urn://Ocpp/Cs/2015/10/}CentralSystemService
from class ocpp.cs._2015._10.CentralSystemService
[... 66 lines cut ...]
WebApplicationContext initialized in 44669 ms
[INFO ] 2023-10-13 07:55:03,477
org.eclipse.jetty.server.handler.ContextHandler.steve (main) -
Initializing Spring DispatcherServlet 'spring-dispatcher'
[INFO ] 2023-10-13 07:55:03,477
org.springframework.web.servlet.DispatcherServlet (main) -
Initializing Servlet 'spring-dispatcher'
[INFO ] 2023-10-13 07:55:03,484
org.springframework.web.servlet.DispatcherServlet (main) - Completed
initialization in 6 ms
[INFO ] 2023-10-13 07:55:03,564
org.eclipse.jetty.server.handler.ContextHandler (main) - Started
***@***.***{/steve,jar:file:/home/tolto/steve/target/steve.jar!/webapp,AVAILABLE}
[INFO ] 2023-10-13 07:55:03,603
org.eclipse.jetty.server.AbstractConnector (main) - Started
***@***.***{HTTP/1.1, (http/1.1)}{0.0.0.0:8080}
[INFO ] 2023-10-13 07:55:03,609 org.eclipse.jetty.server.Server (main)
- Started ***@***.***{STARTING}[10.0.14,sto=5000] @41630ms
[INFO ] 2023-10-13 11:05:06,783
org.apache.http.impl.execchain.RetryExec (qtp9353202-128) - I/O
exception (java.net.SocketException) caught when processing request to
{s}->https://api.github.com:443: Connection reset

[INFO ] 2023-10-13 11:05:06,787 org.apache.http.impl.execchain.RetryExec
(qtp9353202-128) - Retrying request to {s}->https://api.github.com:443"

goekay, 2024-02-02:
```
[INFO ] 2023-10-13 11:05:06,783
org.apache.http.impl.execchain.RetryExec (qtp9353202-128) - I/O
exception (java.net.SocketException) caught when processing request to
{s}->[https://api.github.com:443](https://api.github.com/): Connection reset

[INFO ] 2023-10-13 11:05:06,787 org.apache.http.impl.execchain.RetryExec
(qtp9353202-128) - Retrying request to {s}->[https://api.github.com:443](https://api.github.com/)
```

just catched this last part which might be relevant for identifying the problem: steve cannot call api.github.com and there are exceptions. i think this is a good indicator that your steve instance cannot access internet, and inversely will not be accessible from internet. and hence the stations not being able to connect to your steve.

i am closing this issue due to inactivity, and and my hunch that this is about network issues.

closed 2024-02-02 by goekay