steve-community/steve #1264: Transactions have same start/stop value

andreas-fey, 2023-10-13:
#### Specifications
```
SteVe Version     : 3.6.0 
Operating system  : Docker
Database          : 1.0.3
Wallbox: Fronius Wattpilot, Firmware 40.7
OCPP protocoll: ocpp1.6J
```

#### Expected Behavior
When I go through the list of transactions, the stop value is higher than the start value; substracting the start from the stop value leads to the energy in kWh which was loaded

#### Actual Behavior
When I iterate through the list of transactions, every transaction has the same value for start & stop; i only can manually calculate the loaded energy by comparing transation i-1 with i, i with o+1 and so forth.

#### Steps to Reproduce the Problem
User the wallbox above, open the transaction search

goekay, 2023-10-13:
does the chargebox send stop transaction messages? it sounds like it does not.

andreas-fey, 2023-10-13:
How can I check that?

goekay, 2023-10-13:
the logs. look for `StartTransaction` and `StopTransaction` messages and their contents.

goekay, 2024-05-20:
closing due to inactivity.

closed 2024-05-20 by goekay