steve-community/steve #1271: Cannot change column 'chargeBoxId': used in a foreign key constraint 'stevedb/FK_chargeBoxId_c' of table 'stevedb/connector'

Nexus20, 2023-10-15:
#### Checklist

- I checked other issues already, but found no answer/solution
- I checked the documentation and wiki, but found no answer/solution
- I am running the latest version and the issue still occurs
- I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : ...
Operating system  : Windows 11
JDK               : 18
Database          : MariaDB 10.4
```

#### Expected Behavior

Adequate compile

#### Actual Behavior

org.flywaydb.core.internal.command.DbMigrate$FlywayMigrateException: Migration V0_7_7__update.sql failed
[ERROR] -----------------------------------
[ERROR] SQL State  : HY000
[ERROR] Error Code : 1833
[ERROR] Message    : Cannot change column 'chargeBoxId': used in a foreign key constraint 'stevedb/FK_chargeBoxId_c' of table 'stevedb/connector'
[ERROR] Location   : G:\Универ\Диплом\steve-steve-3.6.0\src\main\resources\db\migration\V0_7_7__update.sql (G:\Универ\Диплом\steve-steve-3.6.0\src\main\resources\db\migration\V0_7_7__update.sql)
[ERROR] Line       : 1
[ERROR] Statement  : ALTER TABLE `chargebox`
[ERROR] CHANGE COLUMN `chargeBoxId` `chargeBoxId` VARCHAR(255) NOT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `endpoint_address` `endpoint_address` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `ocppProtocol` `ocppProtocol` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `chargePointVendor` `chargePointVendor` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `chargePointModel` `chargePointModel` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `chargePointSerialNumber` `chargePointSerialNumber` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `chargeBoxSerialNumber` `chargeBoxSerialNumber` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `fwVersion` `fwVersion` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `fwUpdateStatus` `fwUpdateStatus` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `iccid` `iccid` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `imsi` `imsi` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `meterType` `meterType` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `meterSerialNumber` `meterSerialNumber` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `diagnosticsStatus` `diagnosticsStatus` VARCHAR(255) NULL DEFAULT NULL COMMENT ''

#### Steps to Reproduce the Problem

  1. Download steve
  2. Build Steve

#### Additional context

...

Nexus20, 2023-10-15:
Steve version 3.6.0

goekay, 2023-10-15:
did you check other issues? this was covered. pls pin your mariadb version to one of [these](https://github.com/steve-community/steve/blob/master/.github/workflows/main.yml#L11).

closed 2023-10-15 by goekay