steve-community/steve #1279: Remote Start/Stop transaction

andreschinchilla81, 2023-10-27:
I want to change a part of the code so that when a card is used in the card reader it does not allow the transaction but if the same card is used from the "RemoteStartTransaction" request it does allow it, any ideas?
It's just for internal control

fnkbsi, 2023-10-31:
I don't understand the use case, so my suggestion may not fit.
If you use a made-up OCPP tag, then there will be no card with this UID. No code change required.

andreschinchilla81, 2023-11-07:
i need the card to stop the transaction

sohaibbinmohsin, 2024-01-31:
When you use card through RFID card reader, the charger makes the "Start Transaction" request. What you can do is remove this functionality from the server but keep the "Stop Transaction" functionality. This way when anyone tries to charge using the card, the charger will not allow it but when they want to stop charging, the charger will allow it.

One thing to note is that there is a difference between "Start Transaction" and "Remote Start Transaction":
1. Start/Stop Transaction: Charge Point ---make request to ----> Central System
    Use case: Charging is allowed/initiated through RFID card
2. Remote Start/Stop Transaction: Central System ---make request to ----> Charge point
   Use case: Charging is allowed/initiated through SteVe dashboard

goekay, 2024-08-07:
this is not possible currently in steve. it is also a very specific use case you have. i see little value implementing it globally in steve, even though the implementation would be not too complicated. here are some initial thoughts to get you started.

you basically want to assign 2 boolean properties to each ocpp tag (or as you call, card): `allowedForAuthorize`, `allowedForRemoteStart`. you need to do some database changes, and set these two labels accordingly for your tags. the settings for your use case would be:

| ocpp tag | allowedForAuthorize | allowedForRemoteStart |
|--------|--------|--------|
|tag-1 | false | true |

then you need to change the code path where we evaluate AuthorizeRequest coming from a station and the code path where we prepare RemoteStartTransaction to use properties.

closed 2024-08-07 by goekay