steve-community/steve #1292: Inserting Invalid Timestamp Results in '0000-00-00 00:00:00.000000' and Causes SQL Exception

Tano-Coppoletta, 2023-11-07:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [ ] I am running the latest version and the issue still occurs
- [x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.6.0
Operating system  : Ubuntu 22.04.1
JDK               : openjdk version "11.0.20.1"
Database          : version 1.0.3
```

## Description
When sending a StartTransaction OCPP message with the `timestamp` parameter set to `1000000`, the server incorrectly stores the `start_timestamp` in the database as `0000-00-00 00:00:00.000000`. This leads to an SQL exception when the system tries to retrieve the transaction data.

## Steps to Reproduce
1. Send a StartTransaction OCPP message with `timestamp` set to `1000000`.
2. Observe the `start_timestamp` in the database; note that it is stored as `0000-00-00 00:00:00.000000`.
3. Navigate to the web interface to view the transaction data.
4. Encounter the SQL exception error as shown in the attached screenshot.

## Expected Behavior
The server should handle the `timestamp` value properly by storing a valid timestamp or rejecting the message if the timestamp is invalid.

## Actual Behavior
The server is storing an invalid timestamp which causes an SQL exception when attempting to read the transaction data, as seen in the error message: `Error while reading field: 'stevedb'.'transaction'.'start_timestamp'`.

## Supporting Material
[image]

Tano-Coppoletta, 2023-12-02:
Hello,

I'm reaching out for an update regarding this issue.

We are in the process of submitting Common Vulnerabilities and Exposures (CVEs) related to this matter. However, we can wait if you are in the process of fixing it.

Additionally, we intend to publish our findings in an academic paper.

Your prompt response and attention to this matter would be greatly appreciated as it will greatly inform our next steps.

Thank you and looking forward to your response.

goekay, 2023-12-03:
this is a valid bug in our ocpp-jaxb dependency. the issue was created and fix is already merged: https://github.com/steve-community/ocpp-jaxb/issues/13

thanks @Tano-Coppoletta

commit 00e4b92f (2023-12-03): update ocpp-jaxb (closes #1292)