steve-community/steve #1302: Connector status is not updating

maybeshaed, 2023-11-17:
#### Checklist

- [x ] I checked other issues already, but found no answer/solution
- [x ] I checked the documentation and wiki, but found no answer/solution
- [x ] I am running the latest version and the issue still occurs
- [x ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     :  3.6.0
Operating system  : Debian 11
Database          : mariadb
```

Hello everyone, I discovered an error in the operation of the server, the fact is that when the status of the charging port changes in any way, the status in the database is updated [(screenshot)](https://prnt.sc/8044jfacYXqA), but in the panel and in the API itself it remains the same. [screenshot](https://prnt.sc/IWcK-0gswJkB)

#### Additional context

I`m using user`s branch that allows more API (idk if it`s the reason of trouble)

maybeshaed, 2023-11-19:
UPD:
My DC station totally doesnt works with status
Here i reserved it 
[INFO ] 2023-11-19 20:38:15,255 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger (qtp1706099897-2291) - [chargeBoxId=EV0070, sessionId=c5136ce7-599c-d6ee-cc06-15b208f2dc14] Sending: [3,"8DA7EDEA-871B-11EE-871B-11EE95EAB103",{"currentTime":"2023-11-19T20:38:15.253Z"}]
[INFO ] 2023-11-19 20:38:22,016 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger (SteVe-Executor-3) - [chargeBoxId=EV0070, sessionId=c5136ce7-599c-d6ee-cc06-15b208f2dc14] Sending: [2,"bc3840ed-f0f1-4682-99f9-88d0a2858898","ReserveNow",{"connectorId":1,"expiryDate":"2023-11-20T00:00:00.000Z","idTag":"0E4E1EA","reservationId":26}]
[INFO ] 2023-11-19 20:38:23,131 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger (qtp1706099897-2291) - [chargeBoxId=EV0070, sessionId=c5136ce7-599c-d6ee-cc06-15b208f2dc14] Received: [3,"bc3840ed-f0f1-4682-99f9-88d0a2858898",{"status":"Accepted"}]
[INFO ] 2023-11-19 20:38:23,350 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger (qtp1706099897-2272) - [chargeBoxId=EV0070, sessionId=c5136ce7-599c-d6ee-cc06-15b208f2dc14] Received: [2,"947BC2E0-871B-11EE-871B-11EE95EAB103","StatusNotification",{"connectorId":1,"errorCode":"NoError","info":"Plug reserved to a concrete user","status":"Reserved","timestamp":"2023-11-19T20:38:22Z","vendorId":"CIRCONTROL"}]
[INFO ] 2023-11-19 20:38:23,356 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger (qtp1706099897-2272) - [chargeBoxId=EV0070, sessionId=c5136ce7-599c-d6ee-cc06-15b208f2dc14] Sending: [3,"947BC2E0-871B-11EE-871B-11EE95EAB103",{}]
[INFO ] 2023-11-19 20:38:23,541 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger (qtp1706099897-2291) - [chargeBoxId=EV0070, sessionId=c5136ce7-599c-d6ee-cc06-15b208f2dc14] Received: [2,"949A5818-871B-11EE-871B-11EE95EAB103","StatusNotification",{"connectorId":2,"errorCode":"NoError","info":"Socket locked by controlling plug","status":"Unavailable","timestamp":"2023-11-19T20:38:22Z","vendorId":"CIRCONTROL"}]

And connector status is not updating here
[image]

goekay, 2024-02-02:
> Hello everyone, I discovered an error in the operation of the server, the fact is that when the status of the charging port changes in any way, the status in the database is updated [...]

when rendering the status of a connector, steve always selects the status with the latest timestamp. 

pls look at the rows from the database again. you will see that for connector_pks 89 and 90 the status available at 14:07 is the most recent. and the screenshot from web page renders that. why do you think there is a problem?

goekay, 2024-02-02:
> UPD:
> My DC station totally doesnt works with status
> Here i reserved it
> ...

the status values you sent from logs are from 2023-11-19T20:38:22Z. the webpage shows the status from 22:36. notice that 22:36 is more recent than 20:38. apparently, within these 2 hours things happened and the station sent more status updates.

goekay, 2025-03-17:
closing due to inactivity.

closed 2025-03-17 by goekay