steve-community/steve #1305: Possible Non-Compliance with OCPP Specification Regarding Unique Message IDs

Tano-Coppoletta, 2023-11-18:
#### Checklist

- [ ] I checked other issues already, but found no answer/solution
- [ ] I checked the documentation and wiki, but found no answer/solution
- [ ] I am running the latest version and the issue still occurs
- [x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.6.0
Operating system  : Ubuntu 22.04.1
JDK               : openjdk version "11.0.20.1"
Database          : version 1.0.3
```

#### Issue Description
According to the OCPP specification, each `CALL` message's ID must be unique and not reused within the same WebSocket connection. However, during testing, it was possible to send multiple messages using the same message ID without any rejection by the server.

#### Steps to Reproduce
1. Establish a WebSocket connection to the Steve OCPP server.
2. Send multiple OCPP `CALL` messages using the same message ID.
3. Observe that the server accepts and processes these messages without enforcing the uniqueness of message IDs.

#### Expected Behavior
The server should enforce the OCPP specification by checking the uniqueness of the message ID for each `CALL` message and reject any messages with a duplicate message ID.

#### Actual Behavior
The server processes messages with duplicate message IDs, which is against the OCPP specification and could lead to message tracking issues or misinterpretation of message responses.

#### Potential Impact
This behavior could cause confusion in message correlation, response handling, and could potentially impact the transactional integrity of the communication between charging stations and the central system.

#### Suggested Solution
Implement a check on the server to ensure message IDs are unique for each `CALL` message on the same WebSocket connection and reject any messages that violate this rule.

#### Compliance Reference
Please refer to section 4.1.4 of the OCPP JSON 1.6 documentation that states the requirement for message ID uniqueness.

PratikVpel, 2023-11-25:
I agree with this, i've faced a same issue

Tano-Coppoletta, 2023-12-02:
Hello,

I'm reaching out for an update regarding this issue.

We are in the process of submitting Common Vulnerabilities and Exposures (CVEs) related to this matter. However, we can wait if you are in the process of fixing it.

Additionally, we intend to publish our findings in an academic paper.

Your prompt response and attention to this matter would be greatly appreciated as it will greatly inform our next steps.

Thank you and looking forward to your response.