steve-community/steve #1309: When station complete dead status

maybeshaed, 2023-11-20:
#### Checklist

- [x ] I checked other issues already, but found no answer/solution
- [x ] I checked the documentation and wiki, but found no answer/solution
- [x ] I am running the latest version and the issue still occurs
- [x ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : Latest
Operating system  : Debian
Database          :  mariadb
```

Hi there! In case of completly dead station (when heartbeat doesnt recieved caused error of sim and station cant be pinged 
[image]
I want to got from steve an JsonAndDisconnected status true, as other stations that are not available right now. Now im getting status jsonAndDisconnected: false, it shows me that station is active and data are confusing. Thank you!
[image]

goekay, 2023-12-03:
are you using a modified version of steve? because steve does not expose a rest endpoint as you showed in the second screenshot

maybeshaed, 2023-12-03:
Hi @goekay !
Thank you for reply, yes i`m using pull request provided by one of the steve`s users.
https://github.com/steve-community/steve/pull/1291

fnkbsi, 2023-12-05:
The modified version supports both ConnectorStatusFilter methods. Are there different results on the web-site when changing the filter? (Does the error only occur if  filter PreferOthersWithStatusOfZero is choosen?)

fnkbsi, 2023-12-06:
Does the log include any error messages in context with this connection or the chargebox? (Is the timing also the hour?)

goekay, 2025-08-29:
closing this one due to inactivity.