steve-community/steve #1316: Maven wrapper now found during K8s docker build

ml170722d, 2023-11-29:
The image for k8s can not be build because the Maven wrapper is not found.

Build fails on this line.
https://github.com/steve-community/steve/blob/6cdc22bfa056f5d21d6513db7104ce057607148a/k8s/docker/Dockerfile#L24

The issue is that the `.mvn` directory and `mvnw` executable are not added to the docker image during the build.

goekay, 2024-01-07:
@ml170722d can you briefly explain how you are using the k8s with steve?

background: https://github.com/steve-community/steve/issues/1351

ml170722d, 2024-01-07:
Hi,

I was trying to run steve in my cluster for testing porpoises, but was unable to build the Docker image. The cause was mentioned above.

When I added 1 or 2 lines of code to Dockerfile it worked end I continued to use steve normally. There was nothing special about steve setup and usage. Lines I added were for adding `.mvn` directory with wrapper inside.

Basically, I was using steve as a plug in and play solution for my tests. Hope this explanation is good enough.

closed 2024-05-20 by goekay