steve-community/steve #1323: NotImplemented The action 'RemotStartTransaction' you are looking for is not found

yuhui6688, 2023-12-05:
#### Checklist

- [ ] I checked other issues already and found no answer
- [ ] I checked the documentation and found no answer
- [ ] I am running the latest version and the feature i am requesting is not implemented there

#### Describe the problem you are trying to solve

I have successfully deployed the Steve server and attempted to connect using websocket, but received an error message when sending the Remote Start Transaction. I am not sure why, please advise. Thank you very much!

ERROR Code:
 [4,"4","NotImplemented","The action 'RemotStartTransaction' you are looking for is not found",{}]
...

#### Describe the solution you'd like

...

#### Describe alternative solutions or features you've considered 

...

#### Additional context

...

yuhui6688, 2023-12-05:
This is a partial sending code：
[image]

juherr, 2023-12-05:
The remote start message is from CSMS to the station and not from the station to the CSMS.

What are you trying to do?

juherr, 2023-12-06:
What is the simulated App you refer?

I think you want to make a StartTransaction:
[image]

Have a look at #1000 for the API subject.

yuhui6688, 2023-12-06:
As shown in the picture below, this is the goal I want to achieve. I don’t know how to do it. Please give me your advice. Thank you very much!
[image]

yuhui6688, 2023-12-06:
Assuming that the ChangePoint status is online, should the App send instructions directly to the Central System or ChangePoint?

goekay, 2023-12-06:
> should the App send instructions directly to the Central System or ChangePoint?

to the central system, i.e. steve. steve is responsible for all bidirectional communication with charge points.

goekay, 2023-12-06:
> Does your explanation mean that the APP sends the StartTransaction command to Steve, and then Steve sends the command to the ChargePoint to start the charging function?

yes. but, the api call  "app -> steve" will be an independent one. it is not the StartTransaction or RemoteStartTransaction of ocpp. you can define it as you want, including the payload. steve would need to understand it, map it to a RemoteStartTransaction and send this to the station.

> How does the app know whether it is successful or not? 

the api call "app -> steve" will be waiting for the response from station to arrive at steve. steve will map it to a response payload that you specify and return it. 

something like this:

[image]

yuhui6688, 2023-12-06:
Does Steve have this API, or do you need to implement the service yourself? Thank you so much！

goekay, 2023-12-06:
it does not, and hence the issue reference above from @juherr

yuhui6688, 2023-12-06:
@jpires If Steve does not have a relevant API, how can I implement this service interface based on Steve's source code? Thank you so much!

yuhui6688, 2023-12-07:
@juherr Your reply has given me great encouragement and help! Thank you very much!
I have modified Steve's code according to your method and compiled it successfully, but when I browsed in chrome, I got an error:
[image]
Check the views folder that is not generated at all in the project.
[image]
 I think I am in trouble. I have tried many methods to no avail. Can you help me?

yuhui6688, 2023-12-07:
OK,Thanks very much!

fnkbsi, 2023-12-07:
The compiled jsp-Files can found under "\target\classes\org\apache\jsp\WEB_002dINF\views". As I see in the picture, your folder org is missing. Have you tried a clean recompiling (mvn clean package)?