steve-community/steve #1328: Where is the log folder / files using docker-compose?

jot4p, 2023-12-12:
#### Checklist

- [ x] I checked other issues already, but found no answer/solution
- [ x] I checked the documentation and wiki, but found no answer/solution
- [ x] I am running the latest version and the issue still occurs
- [ ]x I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : docker-compose
Operating system  : ubuntu

#### Expected Behavior
where is the folder ?

juherr, 2023-12-12:
Please, explain the question/issue.

goekay, 2023-12-12:
docker-compose uses [the docker application profile](https://github.com/steve-community/steve/blob/master/Dockerfile#L24) which activates [these configs](https://github.com/steve-community/steve/tree/master/src/main/resources/config/docker) which means [CONSOLE as logger](https://github.com/steve-community/steve/blob/master/src/main/resources/config/docker/log4j2.xml#L9-L11).

there is no log file. we log to stdout.

jot4p, 2023-12-12:
Ok, maybe is more easy to expose my problem..

I'm trying to add a new "Z-BOX charger", very similar to BENY charger:
https://www.beny.com/product-item/22kw-dual-socket-smart-ev-charger/

 I configured ocpp 1.6j for steve and receive on the screen : "starting session on the server with SUC" 
 
The connection is established but I don't see any "new" device in my list of chargers (unknown list) .. 

Any Idea ?  

LOG in portainer:  This is the full log ? 

[image]

goekay, 2023-12-13:
your problems seem to be specific to you and your station. sorry but this has nothing to do with steve. github issues are not a platform for customer service. closing the issue.

closed 2023-12-13 by goekay