steve-community/steve #1329: Build succeeds but jar file runs on localhost, not on IP address

MadLamprey, 2023-12-13:
Starting. FAILED!
Please refer to the log file for details
Exception in thread "main" java.util.concurrent.ExecutionException: java.lang.IllegalStateException: STOPPED
        at java.base/java.util.concurrent.CompletableFuture.reportGet(CompletableFuture.java:395)
        at java.base/java.util.concurrent.CompletableFuture.get(CompletableFuture.java:2022)
        at org.eclipse.jetty.server.Server.doStop(Server.java:490)
        at org.eclipse.jetty.util.component.AbstractLifeCycle.stop(AbstractLifeCycle.java:132)
        at de.rwth.idsg.steve.JettyServer.stop(JettyServer.java:185)
        at de.rwth.idsg.steve.SteveProdStarter.stop(SteveProdStarter.java:80)
        at de.rwth.idsg.steve.Application.stop(Application.java:85)
        at de.rwth.idsg.steve.Application.main(Application.java:68)
Caused by: java.lang.IllegalStateException: STOPPED
        at org.eclipse.jetty.server.handler.ContextHandler.shutdown(ContextHandler.java:774)
        at java.base/java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:195)
        at java.base/java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1655)
        at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:484)
        at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:474)
        at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:550)
        at java.base/java.util.stream.AbstractPipeline.evaluateToArrayNode(AbstractPipeline.java:260)
        at java.base/java.util.stream.ReferencePipeline.toArray(ReferencePipeline.java:517)
        at org.eclipse.jetty.util.component.Graceful.shutdown(Graceful.java:146)
        ... 6 more

goekay, 2023-12-13:
no context, no info, not a proper issue.

closed 2023-12-13 by goekay

MadLamprey, 2023-12-13:
The error message I received in console is:

Starting. FAILED!
Please refer to the log file for details
Exception in thread "main" java.util.concurrent.ExecutionException: java.lang.IllegalStateException: STOPPED
at java.base/java.util.concurrent.CompletableFuture.reportGet(CompletableFuture.java:395)
at java.base/java.util.concurrent.CompletableFuture.get(CompletableFuture.java:2022)
at org.eclipse.jetty.server.Server.doStop(Server.java:490)
at org.eclipse.jetty.util.component.AbstractLifeCycle.stop(AbstractLifeCycle.java:132)
at de.rwth.idsg.steve.JettyServer.stop(JettyServer.java:185)
at de.rwth.idsg.steve.SteveProdStarter.stop(SteveProdStarter.java:80)
at de.rwth.idsg.steve.Application.stop(Application.java:85)
at de.rwth.idsg.steve.Application.main(Application.java:68)
Caused by: java.lang.IllegalStateException: STOPPED
at org.eclipse.jetty.server.handler.ContextHandler.shutdown(ContextHandler.java:774)
at java.base/java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:195)
at java.base/java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1655)
at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:484)
at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:474)
at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:550)
at java.base/java.util.stream.AbstractPipeline.evaluateToArrayNode(AbstractPipeline.java:260)
at java.base/java.util.stream.ReferencePipeline.toArray(ReferencePipeline.java:517)
at org.eclipse.jetty.util.component.Graceful.shutdown(Graceful.java:146)
... 6 more

When the server.host is set to 127.0.0.1, the build succeeds and the jar file runs. However, when the server.host is set to my IP address, the build succeeds but the jar file run fails and I get the above error message.

MadLamprey, 2023-12-13:
@goekay Please if you could offer some guidance.

goekay, 2023-12-13:
it looks like the server cannot bind to your external IP. there might be various reasons for this. it is a networking and configuration issue. as you see, this is specific to you and your situation and not steve. github issues are not a platform for customer service.