steve-community/steve #1347: Steve communication with Alfen Eve Charger broken

wrietv, 2023-12-31:
#### Checklist

- [ x] I checked other issues already, but found no answer/solution
- [ x] I checked the documentation and wiki, but found no answer/solution
- [ x] I am running the latest version and the issue still occurs
- [ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.6.0
Operating system  : Raspbian bullseye
JDK               : openjdk 11.0.21
Database          : Mariadb ver 15.1..
```

#### Expected Behavior
Untill the last firmware update Steve worked flawlessly with the Alfen Eve Charger

#### Actual Behavior
After the last firmware update of the Alfen Eve Charger to version 6.5.0-4217, the communication stops with the following problem in the Steve logfile:

[INFO ] 2023-12-29 12:14:04,509 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=TRAFFIC_0175, sessionId=23ff971b-3f54-cc00-ad75-0ecce5d909f9] Sending: [4,"7","FormationViolation","The payload for action could not be deserialized",{"errorMsg":"Unrecognized field \"values\" (class ocpp.cs._2012._06.MeterValue), not marked as ignorable (2 known p..."}]

#### Steps to Reproduce the Problem

  1. Connect Steve to the Alfen with the latest firmware. 

#### Additional context
I have down graded Steve to 3.5 and 3.4 to no avail. Downgrading of the Alfen firmware is not possible.

juherr, 2024-01-01:
The receiving payload is missing. Could you add it?

wrietv, 2024-01-01:
Not sure what you mean by that, but I have the Steve log and the Alfen Eve log available.
In the Alfen Eve log I find the following:
2023-12-31T14:53:12.289Z:USER:comConnect.c:328:wired connect
2023-12-31T14:53:12.621Z:INFO:websockets.c:445:Connected to Socket 192.168.1.214:8080
2023-12-31T14:53:12.632Z:INFO:websockets.c:2202:WS timeout: 10000 ms
2023-12-31T14:53:13.050Z:INFO:websockets.c:813:WebSocket upgrade successful
2023-12-31T14:53:13.070Z:USER:taskOCPP.c:477:Back office connection available
2023-12-31T14:53:18.082Z:COM:ocpp_rpc.c:681:-> [2,"1","BootNotification",{"chargePointVendor":"Alfen BV","chargePointModel":
2023-12-31T14:53:18.082Z:COM:ocpp_rpc.c:681:"NG910-60023","chargePointSerialNumber":"ACE0177305","chargeBoxSerialNumber":"TR
2023-12-31T14:53:18.082Z:COM:ocpp_rpc.c:681:AFFIC_0175","firmwareVersion":"6.5.0-4217"}]
2023-12-31T14:53:18.531Z:COM:ocpp_rpc.c:1127:<- [3,"1",{"status":"Accepted","currentTime":"2023-12-31T14:53:19.472Z","heartbe
2023-12-31T14:53:18.531Z:COM:ocpp_rpc.c:1127:atInterval":60}]
2023-12-31T14:53:28.578Z:COM:ocpp_rpc.c:681:-> [2,"2","StartTransaction",{"timestamp":"2023-12-31T11:47:54Z","connectorId":1
2023-12-31T14:53:28.578Z:COM:ocpp_rpc.c:681:,"meterStart":2145278,"idTag":"ACE0177305"}]
2023-12-31T14:53:29.265Z:COM:ocpp_rpc.c:1127:<- [3,"2",{"transactionId":2,"idTagInfo":{"status":"Blocked"}}]
2023-12-31T14:53:29.277Z:INFO:db_transaction.c:16:Updated TransactionStart for socket 1 at 0x000009ae (ts: 2023-12-31 11:47:54)
2023-12-31T14:53:29.289Z:INFO:taskMaster.c:2374:Socket #1: tag marked invalid
2023-12-31T14:53:29.300Z:INFO:db_transaction.c:16:Updated TransactionStop for socket 1 at 0x00000a0c (ts: 2023-12-31 11:50:33)
2023-12-31T14:53:29.316Z:COM:ocpp_rpc.c:681:-> [2,"3","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"
2023-12-31T14:53:29.316Z:COM:ocpp_rpc.c:681:Available"}]
2023-12-31T14:53:29.457Z:COM:ocpp_rpc.c:1127:<- [3,"3",{}]
2023-12-31T14:53:29.464Z:COM:ocpp_rpc.c:681:-> [2,"4","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"
2023-12-31T14:53:29.464Z:COM:ocpp_rpc.c:681:Available"}]
2023-12-31T14:53:29.554Z:COM:ocpp_rpc.c:1127:<- [3,"4",{}]
2023-12-31T14:53:30.328Z:COM:ocpp_rpc.c:681:-> [2,"5","MeterValues",{"connectorId":1,"transactionId":2,"values":[{"timestamp
2023-12-31T14:53:30.328Z:COM:ocpp_rpc.c:681:":"2023-12-31T11:47:54Z","values":[{"value":"2145.278","context":"Transaction.Be
2023-12-31T14:53:30.328Z:COM:ocpp_rpc.c:681:gin","unit":"kWh"}]}]}]
2023-12-31T14:53:30.597Z:COM:ocpp_rpc.c:1127:<- [3,"5",{}]
2023-12-31T14:53:31.632Z:COM:ocpp_rpc.c:681:-> [2,"6","MeterValues",{"connectorId":1,"transactionId":2,"values":[{"timestamp
2023-12-31T14:53:31.632Z:COM:ocpp_rpc.c:681:":"2023-12-31T11:50:33Z","values":[{"value":"2145.539","context":"Transaction.En
2023-12-31T14:53:31.632Z:COM:ocpp_rpc.c:681:d","unit":"kWh"}]}]}]
2023-12-31T14:53:31.746Z:COM:ocpp_rpc.c:1127:<- [3,"6",{}]
2023-12-31T14:53:32.765Z:COM:ocpp_rpc.c:681:-> [2,"7","StopTransaction",{"timestamp":"2023-12-31T11:50:33Z","transactionId":
2023-12-31T14:53:32.765Z:COM:ocpp_rpc.c:681:2,"meterStop":2145539,"idTag":"ACE0177305","transactionData":[{"values":[{"times
2023-12-31T14:53:32.765Z:COM:ocpp_rpc.c:681:tamp":"2023-12-31T11:47:54Z","values":[{"value":"2145.278","context":"Transactio
2023-12-31T14:53:32.765Z:COM:ocpp_rpc.c:681:n.Begin","unit":"kWh"}]},{"timestamp":"2023-12-31T11:50:33Z","values":[{"value":
2023-12-31T14:53:32.765Z:COM:ocpp_rpc.c:681:"2145.539","context":"Transaction.End","unit":"kWh"}]}]}]}]
2023-12-31T14:53:32.902Z:COM:ocpp_rpc.c:1127:<- [4,"7","FormationViolation","The payload for action could not be deserialized
2023-12-31T14:53:32.902Z:COM:ocpp_rpc.c:1127:",{"errorMsg":"Unrecognized field \"values\" (class ocpp.cs._2012._06.MeterValue
2023-12-31T14:53:32.902Z:COM:ocpp_rpc.c:1127:), not marked as ignorable (2 known p..."}]
2023-12-31T14:53:32.945Z:ERROR:ocpp_json.c:462:WAMP_ERROR on (FormationViolation) Desc: The payload for action could not be des
2023-12-31T14:53:32.945Z:ERROR:ocpp_json.c:462:erialized
2023-12-31T14:53:32.964Z:ERROR:ocpp_rpc.c:438:reply error REPLY ERROR
2023-12-31T14:54:32.988Z:COM:ocpp_rpc.c:681:-> [2,"8","StopTransaction",{"timestamp":"2023-12-31T11:50:33Z","transactionId":
2023-12-31T14:54:32.988Z:COM:ocpp_rpc.c:681:2,"meterStop":2145539,"idTag":"ACE0177305","transactionData":[{"values":[{"times
2023-12-31T14:54:32.988Z:COM:ocpp_rpc.c:681:tamp":"2023-12-31T11:47:54Z","values":[{"value":"2145.278","context":"Transactio
2023-12-31T14:54:32.988Z:COM:ocpp_rpc.c:681:n.Begin","unit":"kWh"}]},{"timestamp":"2023-12-31T11:50:33Z","values":[{"value":
2023-12-31T14:54:32.988Z:COM:ocpp_rpc.c:681:"2145.539","context":"Transaction.End","unit":"kWh"}]}]}]}]
2023-12-31T14:54:33.113Z:COM:ocpp_rpc.c:1127:<- [4,"8","FormationViolation","The payload for action could not be deserialized
2023-12-31T14:54:33.113Z:COM:ocpp_rpc.c:1127:",{"errorMsg":"Unrecognized field \"values\" (class ocpp.cs._2012._06.MeterValue
2023-12-31T14:54:33.113Z:COM:ocpp_rpc.c:1127:), not marked as ignorable (2 known p..."}]
2023-12-31T14:54:33.152Z:ERROR:ocpp_json.c:462:WAMP_ERROR on (FormationViolation) Desc: The payload for action could not be des
2023-12-31T14:54:33.152Z:ERROR:ocpp_json.c:462:erialized
2023-12-31T14:54:33.171Z:ERROR:ocpp_rpc.c:438:reply error REPLY ERROR

wrietv, 2024-01-01:
Thanks for your reply. I am not sure what you mean, but attached you find a typical Steve log when the error occurs. In Github I have also posted a pertaining log file from the Alfen Eve.

[INFO ] 2023-12-31 15:53:14,220 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=TRAFFIC_0175, sessionId=89266e52-0694-69fc-936c-798b9f59a188] Connection is established
[INFO ] 2023-12-31 15:53:19,375 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=TRAFFIC_0175, sessionId=89266e52-0694-69fc-936c-798b9f59a188] Received: [2,"1","BootNotification",{"chargePointVendor":"Alfen BV","chargePointModel":"NG910-60023","chargePointSerialNumber":"ACE0177305","chargeBoxSerialNumber":"TRAFFIC_0175","firmwareVersion":"6.5.0-4217"}]
[INFO ] 2023-12-31 15:53:19,472 de.rwth.idsg.steve.service.CentralSystemService16_Service - The boot of the chargebox 'TRAFFIC_0175' with registration status 'Optional[ACCEPTED]' is acknowledged.
[INFO ] 2023-12-31 15:53:19,642 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=TRAFFIC_0175, sessionId=89266e52-0694-69fc-936c-798b9f59a188] Sending: [3,"1",{"status":"Accepted","currentTime":"2023-12-31T14:53:19.472Z","heartbeatInterval":60}]
[INFO ] 2023-12-31 15:53:29,887 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=TRAFFIC_0175, sessionId=89266e52-0694-69fc-936c-798b9f59a188] Received: [2,"2","StartTransaction",{"timestamp":"2023-12-31T11:47:54Z","connectorId":1,"meterStart":2145278,"idTag":"ACE0177305"}]
[ERROR] 2023-12-31 15:53:30,004 de.rwth.idsg.steve.service.OcppTagService - The user with idTag 'ACE0177305' is BLOCKED.
[INFO ] 2023-12-31 15:53:30,139 de.rwth.idsg.steve.repository.impl.OcppServerRepositoryImpl - The connector TRAFFIC_0175/1 is NEW, and inserted into DB.
[INFO ] 2023-12-31 15:53:30,390 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=TRAFFIC_0175, sessionId=89266e52-0694-69fc-936c-798b9f59a188] Sending: [3,"2",{"transactionId":2,"idTagInfo":{"status":"Blocked"}}]
[INFO ] 2023-12-31 15:53:30,500 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=TRAFFIC_0175, sessionId=89266e52-0694-69fc-936c-798b9f59a188] Received: [2,"3","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Available"}]
[INFO ] 2023-12-31 15:53:30,546 de.rwth.idsg.steve.repository.impl.OcppServerRepositoryImpl - The connector TRAFFIC_0175/0 is NEW, and inserted into DB.
[INFO ] 2023-12-31 15:53:30,573 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=TRAFFIC_0175, sessionId=89266e52-0694-69fc-936c-798b9f59a188] Sending: [3,"3",{}]
[INFO ] 2023-12-31 15:53:30,648 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=TRAFFIC_0175, sessionId=89266e52-0694-69fc-936c-798b9f59a188] Received: [2,"4","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Available"}]
[INFO ] 2023-12-31 15:53:30,674 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=TRAFFIC_0175, sessionId=89266e52-0694-69fc-936c-798b9f59a188] Sending: [3,"4",{}]
[INFO ] 2023-12-31 15:53:31,525 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=TRAFFIC_0175, sessionId=89266e52-0694-69fc-936c-798b9f59a188] Received: [2,"5","MeterValues",{"connectorId":1,"transactionId":2,"values":[{"timestamp":"2023-12-31T11:47:54Z","values":[{"value":"2145.278","context":"Transaction.Begin","unit":"kWh"}]}]}]
[INFO ] 2023-12-31 15:53:31,721 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=TRAFFIC_0175, sessionId=89266e52-0694-69fc-936c-798b9f59a188] Sending: [3,"5",{}]
[INFO ] 2023-12-31 15:53:32,827 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=TRAFFIC_0175, sessionId=89266e52-0694-69fc-936c-798b9f59a188] Received: [2,"6","MeterValues",{"connectorId":1,"transactionId":2,"values":[{"timestamp":"2023-12-31T11:50:33Z","values":[{"value":"2145.539","context":"Transaction.End","unit":"kWh"}]}]}]
[INFO ] 2023-12-31 15:53:32,870 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=TRAFFIC_0175, sessionId=89266e52-0694-69fc-936c-798b9f59a188] Sending: [3,"6",{}]
[INFO ] 2023-12-31 15:53:33,970 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=TRAFFIC_0175, sessionId=89266e52-0694-69fc-936c-798b9f59a188] Received: [2,"7","StopTransaction",{"timestamp":"2023-12-31T11:50:33Z","transactionId":2,"meterStop":2145539,"idTag":"ACE0177305","transactionData":[{"values":[{"timestamp":"2023-12-31T11:47:54Z","values":[{"value":"2145.278","context":"Transaction.Begin","unit":"kWh"}]},{"timestamp":"2023-12-31T11:50:33Z","values":[{"value":"2145.539","context":"Transaction.End","unit":"kWh"}]}]}]}]
[ERROR] 2023-12-31 15:53:34,012 de.rwth.idsg.steve.ocpp.ws.pipeline.Deserializer - Exception occurred
com.fasterxml.jackson.databind.exc.UnrecognizedPropertyException: Unrecognized field "values" (class ocpp.cs._2012._06.MeterValue), not marked as ignorable (2 known properties: "value", "timestamp"])
 at [Source: UNKNOWN; byte offset: #UNKNOWN] (through reference chain: ocpp.cs._2012._06.StopTransactionRequest["transactionData"]->java.util.ArrayList[0]->ocpp.cs._2012._06.TransactionData["values"]->java.util.ArrayList[0]->ocpp.cs._2012._06.MeterValue["values"])
	at com.fasterxml.jackson.databind.exc.UnrecognizedPropertyException.from(UnrecognizedPropertyException.java:61) ~[jackson-databind-2.13.2.2.jar:2.13.2.2]
	at com.fasterxml.jackson.databind.DeserializationContext.handleUnknownProperty(DeserializationContext.java:1127) ~[jackson-databind-2.13.2.2.jar:2.13.2.2]
[... cut ...]
	at org.eclipse.jetty.util.thread.strategy.AdaptiveExecutionStrategy.lambda$new$0(AdaptiveExecutionStrategy.java:138) ~[jetty-util-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.util.thread.ReservedThreadExecutor$ReservedThread.run(ReservedThreadExecutor.java:407) ~[jetty-util-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:894) ~[jetty-util-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.util.thread.QueuedThreadPool$Runner.run(QueuedThreadPool.java:1038) ~[jetty-util-10.0.7.jar:10.0.7]
	at java.lang.Thread.run(Thread.java:829) ~[?:?]

juherr, 2024-01-01:
The failing request is:

```json
 [2,"7","StopTransaction",{"timestamp":"2023-12-31T11:50:33Z","transactionId":2,"meterStop":2145539,"idTag":"ACE0177305","transactionData":[{"values":[{"timestamp":"2023-12-31T11:47:54Z","values":[{"value":"2145.278","context":"Transaction.Begin","unit":"kWh"}]},{"timestamp":"2023-12-31T11:50:33Z","values":[{"value":"2145.539","context":"Transaction.End","unit":"kWh"}]}]}]}]
```

Need to check if the payload is legit.

wrietv, 2024-01-01:
I do not have more information now. But I can easily replicate if that helps and send you the logs of that. Strange thing is that Steve communicated fine with Alfen Eve before the firmware update on Alfen.

goekay, 2024-01-01:
the first occurrence of `values` should have been `sampledValue`. this is according to spec.

wrietv, 2024-01-02:
You are undoubtedly right, but how can I resolve that on my side?
Happy New Year!

goekay, 2024-01-02:
> how can I resolve that on my side?

you can go back to an earlier firmware, or contact the manufacturer about this regression.

wrietv, 2024-01-02:
Thanks. The problem is that going back to an earlier firmware has been made impossible by Alfen. So, if that is the only solution, I am probably forced to say goodbye to Steve, to my great regret, because I have fallen in love with it.
Is there nothing that can be done on the Steve side? I will not be the only one with this problem.

juherr, 2024-01-02:
Create an issue on alfen helpdesk. They are reactive and often provide early access firmware with fix.

wrietv, 2024-01-02:
Thanks, I will try.

goekay, 2024-01-02:
> ... I am probably forced to say goodbye to Steve, to my great regret, because I have fallen in love with it.

thanks for the kind words. 

> Is there nothing that can be done on the Steve side?

of course there is. this is software code at the end of the day. i can "bend" the implementation to add some flexibility. i did some similar things in the past for some weird manufacturer decisions and mistakes. but, i dont know how i feel about this after 10 years. i'd rather prefer them doing the right thing, instead of me creating a workaround for their regression.

wrietv, 2024-01-02:
I understand. I will see what I can do involving the manufacturer, but previous encounters as to software issues don't make me very optimistic...

wrietv, 2024-01-03:
I succeeded in acquiring a backlevel firmware for my Alfen Eve. I could install it and now Steve communicates again with the charger as before. Although this is an acceptable workaround, of course it is not a solution. For now I am good, but in the future it will be difficult to maintain.
Anyway all: thanks for your support and have a very good year!

closed 2024-01-07 by goekay