steve-community/steve #1351: Community feedback wanted: Drop kubernetes profile and resources

goekay, 2024-01-07:
our kubernetes configuration comes with the following resources:
- [Dockerfile](https://github.com/steve-community/steve/blob/master/k8s/docker/Dockerfile)
- [Deployment and service manifests](https://github.com/steve-community/steve/tree/master/k8s/yaml)
- [App properties and logging config](https://github.com/steve-community/steve/tree/master/src/main/resources/config/kubernetes)

the Dockerfile and the k8s manifests are quite dull and straightforward. i would even argue that the k8s manifests are not useful, because they probably dont align with your cluster topology/config, and deployment strategies. and hence, i am thinking that the k8s manifests are _context-dependent_ and steve, as a project, cannot give you what you want out of the box. creating these manifests for your cluster should not be that hard for somewhat experienced developer.

therefore, i am questioning whether we should drop them. what do you think? anyone found these useful?

juherr, 2024-01-07:
Hi,

In my opinion, only the prod profile should stay and the test profile should be moved under the test directory.

The dev profile depend on the way each developer will work.

Docker is just a way to run and the docker profile should use the same environment variables than the prod profile.
At the opposite, a docker compose file is maybe missing and could help to run the project from the sources with only on command.

ml170722d, 2024-01-07:
Hello,

To be honest, it is not a big problem if the Dockerfile is dropped or not.
I just pointed out that the provided files in the project do not work out of the box.

I agree that someone with experience can write everything that they need but, it still takes (some) time to do so.
If the project has some of that work done, that decreases the work one must invest in setting up the k8s manifests and Dockerfile.

The project I was working on needed steve temporarily to test my application(s). Because I had most of the work done inside the project, I did not have to spend any time setting everything up. Instead, I had to do a little bit of debugging of Dockerfile. Afterward, I just reported the issue because I thought it would benefit the project.

In my humble opinion, I think that the files mentioned above should not be dropped. They make someone's life a bit easier when using this project for implementation and testing their applications.

goekay, 2024-01-09:
> At the opposite, a docker compose file is maybe missing and could help to run the project from the sources with only on command

what do you mean? we have [this](https://github.com/steve-community/steve/blob/master/docker-compose.yml) already

juherr, 2024-01-09:
Oops, forget my point about compose

goekay, 2024-02-18:
ok, we can and will keep it. thanks for the input.

closed 2024-02-18 by goekay