steve-community/steve #1352: OFFlINE Txdefault charging profile

LydiaBenaida, 2024-01-11:
Hello everyone, 
i declared my charging profiles , have both txdefault and tx , 

1. When Online verything is Ok: when nothing teh EV takes the TXdefault and when i do TX it takes this last one 
2. My question is when i'm OFFLINE my EV doesn't take TxDefault 

Exemple 
here is my decalred charging profile
[image]
But offline it doesnt work ! *
do i need to change this configuration or what !
...

#### Describe alternative solutions or features you've considered 

...

#### Additional context

...

fnkbsi, 2024-01-12:
Would you please explain what you mean with online/offline in this scenario?

The picture shows the website for defining the profile to store in the database.
To send a stored profile to the EVSE use the operations websites (with the ocpp version your using to communicate with the EVSE).
If it's not successful the log provides normally some hints.

LydiaBenaida, 2024-01-12:
Thank you for your answer , 
So by offline i mean a charge point doesn't appear in Steve , it s confugured( leans i already sent a Tx_Default_Profile but not online , example : losing internet connections are malfunctioning.
thank you

fnkbsi, 2024-01-12:
I'm not sure I have understand the behavior correctly. So I try to rephrase it:
A TxDefault profile is send to the charge point (cp). The response of the cp responses with accepted (task website).
Case 1: If the cp is connected, the profile is applied by the cp.
Case 2: If the cp then loses the connection the profile is ignored by the cp.
 
Is this the behavior?

goekay, 2024-01-13:
as far as i understand, the messages are sent and the station accepts these profiles (as response). and it all works as long as the station stays online. if, however, the station becomes offline _later_, the sent and accepted profiles are not used/respected by the offline station.

if this is the situation: we cannot help you. it is the charging station that has a weird behaviour.