steve-community/steve #1353: Add support for SignedUpdateFirmware

astrand, 2024-01-11:
1.6 whitepaper mandates the new SignedUpdateFirmware request, and it is required not to support the old UpdateFirmware. Thus, for a CS supporting the signed version, Steve cannot be used to update firmware.

In other words, it would be good if we could add support for SignedUpdateFirmware.

goekay, 2024-01-17:
i assume you are referencing [OCPP 1.6 Security Whitepaper (3rd edition)](https://www.openchargealliance.org/uploads/files/OCPP-1.6-security-whitepaper-edition-3.zip). none of the enhancements/adjustments mentioned there are implemented in steve.

so, i agree, but not only that. we should aim to implement all the ideas by the whitepaper.

zentim, 2024-03-01:
The whitepaper link has changed, you can find it through this link [OCPP-1.6-security-whitepaper-edition-3-2.zip](https://openchargealliance.org/wp-content/uploads/2023/11/OCPP-1.6-security-whitepaper-edition-3-2.zip) or this page [INFO & WHITEPAPERS](https://openchargealliance.org/whitepapers/).

juherr, 2025-10-28:
@razvanphp They are indeed:

[image]

Which OCTT test case are you referring to exactly?  
https://openchargealliance.org/wp-content/uploads/2025/09/CompliancyTestTool-TestCaseDocument.pdf

juherr, 2025-10-29:
@razvanphp It makes sense that the 1.6 core test cases validate the initial command only. The challenge comes when you aim to validate both specifications, but so far I haven’t found any certification scheme addressing the security whitepaper. As @astrand pointed out, it is possible to validate a particular firmware version or enforce a specific configuration to manage behaviour expectations.

> Also note that since 2025-08-01, for products in EU which falls under the RED directive, signed firmware update is also required thus the old UpdateFirmware must be disabled. Unfortunately many CSMS systems still don't support signed firmware update 🙄

@astrand thanks for raising that point. I wasn’t aware of it, and from what I hear, manufacturers aren’t proactively mentioning it either, even though many reference ISO 15118‑20. Do you have any links or resources that could help me dig deeper into the implications of RED for EV charging equipment?

astrand, 2025-10-30:
Wrt RED, basically the upcoming Cyber Resilience Act has been introduced in advance for internet connected radio equipment. EU DR 2022/30 "activates" clause 3.3 in the earlier RED directive. In practice, this means that you must consider EN 18031 -1 -2 -3. The first one of these requires a software update mechanism which "shall only install software whose integrity and
authenticity are valid at the time of the installation".

goekay, 2025-11-06:
hey folks, i just wanted to ping that this operation is now implemented in [the respective PR](https://github.com/steve-community/steve/pull/1854). also see the [WIP updates](https://github.com/steve-community/steve/pull/1854#issuecomment-3486157126).

i saw `L01.FR.20` and decided to leave this control to CP. steve will offer both variants, UpdateFirmware and SignedUpdateFirmware, and the user can decide.