steve-community/steve #1354: regarding jar files

smdkhalifa7, 2024-01-12:
#### Checklist

- [ ] I checked other issues already and found no answer
- [ ] I checked the documentation and found no answer
- [ ] I am running the latest version and the feature i am requesting is not implemented there

#### Describe the problem you are trying to solve

I want to make changes in the parameters of the feature Data Transfer,
I need to add new variables to it but I couldn't find the file to make changes.
...

#### Describe the solution you'd like

...

#### Describe alternative solutions or features you've considered 

...

#### Additional context

I need to know where I can find the file to make changes
...

fnkbsi, 2024-01-12:
The websites includes all fields of the message DataTransfer.req as defined in the OCPP specification.
Still want to make changes. 
The correspondent jsp File is src\main\resources\webapp\WEB-INF\views\op-forms\DataTransferForm.jsp and the controller for the site is implemented at src\main\java\de\rwth\idsg\steve\web\controller\Ocpp15Controller.java

goekay, 2024-01-17:
why do you want to leave the ocpp spec? that will create compatibility issues. moreover, https://github.com/steve-community/steve/issues/1362#issuecomment-1896711916

closed 2024-01-17 by goekay