steve-community/steve #1355: Why is the StopTransaction executed successfully but the Tasks still exist? How to delete them

yuhui6688, 2024-01-13:
Web Tasks:
[image]

log for StopTransaction executed successfully :
[image]

fnkbsi, 2024-01-13:
The Task site shows the result of a occp operation like sending a profile or RemoteTransactionStop.
The two tasks in your list, missing the response of the charge point. You get more information by clicking on the Task ID.

If you want to check if a transaction is finished, visit the transaction site. If you can't find the transaction your looking for, set the "Transaction Type" to all.

yuhui6688, 2024-01-15:
> 任务站点显示 occp 操作的结果，例如发送配置文件或 RemoteTransactionStop。 您列表中的两项任务缺少充电点的响应。单击任务 ID 可获取更多信息。
> 
> 如果您想检查交易是否完成，请访问交易网站。如果您找不到所需的交易，请将“交易类型”设置为全部。

> The Task site shows the result of a occp operation like sending a profile or RemoteTransactionStop. The two tasks in your list, missing the response of the charge point. You get more information by [...]

I see,Only considering sending data,Tks!

goekay, 2024-02-18:
i am closing this issue, since it looks like resolved by the answer from @fnkbsi

closed 2024-02-18 by goekay