steve-community/steve #1364: Cannot start transaction when tag already holds an active transaction

Smoerrebroed, 2024-01-19:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [x] I am running the latest version and the issue still occurs
- [x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
While a transaction is already in progress for a given tag, I want to start a second transaction on a different charge point. Tag is configured with Max. Active Transaction Count of -1.

SteVe Version     : 3.6.0
Operating system  : Linux
JDK               : 21.0.1
Database          : MariaDB 11.2.2 
```

#### Expected Behavior

Second transaction starts normally.

#### Actual Behavior

Charge point (SteVe) denies initiation of second transaction; remote starting a transaction is not possible as the tag is not even selectable from the drop-down list.

#### Steps to Reproduce the Problem

  1. Set up OCPP tag with Max. Active Transaction Count > 1
  2. Start a transaction for this tag
  3. Try to start a second transaction (on another charge point) for this tag.

fnkbsi, 2024-01-23:
Hi Smoerrebroed, 

could you please clarify in which case the problem occurs?

1. Max. Active Transaction Count > 1
2. Max. Active Transaction Count == -1
3. both cases

Providing a log may also help isolate the problem.

Smoerrebroed, 2024-01-31:
It's both cases. I don't normally use a limit on the number of transactions (-1), but I also tried with 2 and 10, and it didn't make a difference. Neither is the physical tag accepted at the charge point (when another transaction is already in progress) nor can I remote start an additional transaction for a tag that already has a transaction in progress.

fnkbsi, 2024-01-31:
PR #1378 addresses the missing ocpp_tag on the RemoteStart website, when max transaction count is not reached.

Smoerrebroed, 2024-01-31:
Great, will test it out tomorrow. If getActiveIdTags() is only used for remote start of transactions, a similar change would be needed for a local start of a transaction, no? That failed previously as well when there was already a transaction in progress for the same (physical) tag.

fnkbsi, 2024-02-13:
Hi @Smoerrebroed, did you have time to test? Is the issue resolved or only partly?

Smoerrebroed, 2024-02-14:
Hi @fnkbsi , sorry for the late response (due to some personal issues which caused delays in getting this done). So I did test different cases now

- Initiate transaction #1 via physical tag and transaction #2 via physical tag
- Initiate transaction #1 via physical tag and transaction #2 via remote start (same tag)
- Initiate transaction #1 via remote start and transaction #2 via (same) physical tag
- Initiate transaction #1 via remote start and transaction #2 via remote start (same tag)

and I'm happy to report that all seem to be working with the latest version, so from my side the issue is solved. Thank you very much for your efforts and support!

closed 2024-02-14 by Smoerrebroed