steve-community/steve #1374: Can we specify OCPP tag to a charger?

sohaibbinmohsin, 2024-01-30:
Hello,

This is kind of a feature request.

Currently, when we add ocpp tags on steve, there is no way of specifying on which charger the tag should authorize. (Correct me please if am wrong). This way if we have multiple chargers, any tag can work on any charger.

So is there any way to restrict the ocpp tag to work for a specific charger?

Thank you for the great platform. You guys are doing great work!

juherr, 2024-01-30:
Hi,

Currently all enabled tags are authorized on all stations.

sohaibbinmohsin, 2024-01-31:
Hello,
Thank you for the reply. Is there any plan to integrate this feature(separate tags for every station) in the future?

goekay, 2024-01-31:
not really, not on my part at least, because there a gazillion different flavours of ocpp tag -> station mappings.

LydiaBenaida, 2024-01-31:
i did devlop this feature , i created a new association table user(with unique id tag) and chareg box pk , and i associated every user to a charge box, 
so only that user with the specific idtag can start a transaction on this charge box : 

` public AuthorizeResponse authorize(AuthorizeRequest parameters, String chargeBoxIdentity) {
        if (chargePointUserRepository.isAssociated(chargeBoxIdentity, parameters.getIdTag())) {
            // Get the authorization info of the user
            IdTagInfo idTagInfo = ocppTagService.getIdTagInfo(
                    parameters.getIdTag(),
                    false,
                    () -> new IdTagInfo().withStatus(AuthorizationStatus.INVALID), null
            );

            return new AuthorizeResponse().withIdTagInfo(idTagInfo);
        } else {
            // Get the authorization info of the user
            IdTagInfo idTagInfo = ocppTagService.getIdTagInfo(
                    parameters.getIdTag(),
                    false,
                    () -> new IdTagInfo().withStatus(AuthorizationStatus.INVALID), AuthorizationStatus.INVALID
            );

            return new AuthorizeResponse().withIdTagInfo(idTagInfo);
        }
    }`

closed 2024-01-31 by sohaibbinmohsin