steve-community/steve #1376: Smart Charging Automaticaly

LydiaBenaida, 2024-01-31:
Hello everyone, 
i'm trying to make smart charging , 
the idea is : when i start a transaction i can calculate a charging profile and i send it automatically to my charge box using 
the famous SetChargingProfile already implemented in our system.
everytime i start a transaction i clear all the tx profile on the active transaction and calculate a nex tx default and set all the charge box profiles .
but it didnt work correctly . like in tasks i got no reponse every 5 second teh charge box retries but didnt accept my charge profile
my probleme i think it about synchronisation i will share my code 
```
public StartTransactionResponse startTransaction(StartTransactionRequest parameters, String chargeBoxIdentity) {
        if (chargePointUserRepository.isAssociated(chargeBoxIdentity, parameters.getIdTag())) {
            // Get the authorization info of the user, before making tx changes (will affectAuthorizationStatus)
            IdTagInfo info = ocppTagService.getIdTagInfo(
                    parameters.getIdTag(),
                    true,
                    () -> new IdTagInfo().withStatus(AuthorizationStatus.INVALID), null // IdTagInfo is required
            );

            InsertTransactionParams params =
                    InsertTransactionParams.builder()
                            .chargeBoxId(chargeBoxIdentity)
                            .connectorId(parameters.getConnectorId())
                            .idTag(parameters.getIdTag())
                            .startTimestamp(parameters.getTimestamp())
                            .startMeterValue(Integer.toString(parameters.getMeterStart()))
                            .reservationId(parameters.getReservationId())
                            .eventTimestamp(DateTime.now())
                            .build();

            int transactionId = ocppServerRepository.insertTransaction(params);

            applicationEventPublisher.publishEvent(new OcppTransactionStarted(transactionId, params));
            StartTransactionResponse response = new StartTransactionResponse()
                    .withIdTagInfo(info)
                    .withTransactionId(transactionId);

            // Call startSetProfiles after the transaction has started
            startSetProfiles(chargeBoxIdentity);

            return response;
        }else {
                // If the association is not present, return an "user not allowed" response
                StartTransactionResponse response = new StartTransactionResponse()
                        .withTransactionId(-1) // Use a special value to indicate an error
                        .withIdTagInfo(new IdTagInfo().withStatus(AuthorizationStatus.INVALID));

                return response;
            }

    }
    public void startSetProfiles(String chargeBoxIdentity) {
        int count = smartCharging.countActiveChargeBoxOnPhase(chargeBoxIdentity);
        int profileCode = smartCharging.getProfileOfPdl(chargeBoxIdentity,count);
        int chargingProfilePk = smartCharging.getChargingProfilePK(profileCode, "TRANSACTION");
        smartCharging.countActiveChargeBoxOnPhase(chargeBoxIdentity);
        List<ChargeBoxInfo> chargeBoxInfo = smartCharging.getChargeBoxInfoList();
        // --------- clear old profiles --------------
        stopSetProfiles(chargeBoxIdentity);
        //---------------------------------------------
        for (ChargeBoxInfo chargeBox : chargeBoxInfo) {
            // --------------- Params -----------------
            List<ChargePointSelect> chargePointSelectList = new ArrayList<>();
            ChargePointSelect chargePointSelect = new ChargePointSelect(OcppTransport.JSON, chargeBox.getChargeBoxId());
            chargePointSelectList.add(chargePointSelect);
            // ------------- Set Params ---------------
            SetChargingProfileParams setParams = new SetChargingProfileParams();

            setParams.setConnectorId(Integer.valueOf(chargeBox.getConnector_id()));
            setParams.setChargingProfilePk(chargingProfilePk);
            setParams.setChargePointSelectList(chargePointSelectList);
            //---------------Set Profile---------------
            client16.setChargingProfile(setParams);
        }
    }
    private void stopSetProfiles(String chargeBoxIdentity) {
        smartCharging.countActiveChargeBoxOnPhase(chargeBoxIdentity);
        List<ChargeBoxInfo> chargeBoxInfo = smartCharging.getChargeBoxInfoList();
        //-------------- clear the old profiles-------
        for (ChargeBoxInfo chargeBox : chargeBoxInfo) {

            int chargingProfilePK = chargingProfileImpl.getCurrentFromChargeBoxInfo(chargeBox,"TxProfile");
            // --------------- Params -----------------
            if(chargingProfilePK>0) {
                List<ChargePointSelect> chargePointSelectList = new ArrayList<>();
                ChargePointSelect chargePointSelect = new ChargePointSelect(OcppTransport.JSON, chargeBox.getChargeBoxId());
                chargePointSelectList.add(chargePointSelect);
                // ------------- Set Params ---------------
                ClearChargingProfileParams clearParams = new ClearChargingProfileParams();
                clearParams.setConnectorId(Integer.valueOf(chargeBox.getConnector_id()));
                clearParams.setChargingProfilePk(chargingProfilePK);
                clearParams.setChargePointSelectList(chargePointSelectList);
                clearParams.setChargingProfilePurpose(ChargingProfilePurposeType.TX_PROFILE);
                //---------------Set Profile---------------
                client16.clearChargingProfile(clearParams);
            }
        }
    }
    public StopTransactionResponse stopTransaction(StopTransactionRequest parameters, String chargeBoxIdentity) {
        //--------clear all profile on the station and phase of our charge box stopped
        stopSetProfiles(chargeBoxIdentity);
        //-------------------------------------------
        int transactionId = parameters.getTransactionId();
        String stopReason = parameters.isSetReason() ? parameters.getReason().value() : null;

        // Get the authorization info of the user, before making tx changes (will affectAuthorizationStatus)
        IdTagInfo idTagInfo = ocppTagService.getIdTagInfo(
                parameters.getIdTag(),
                false,
                () -> null,null
        );

        UpdateTransactionParams params =
                UpdateTransactionParams.builder()
                                       .chargeBoxId(chargeBoxIdentity)
                                       .transactionId(transactionId)
                                       .stopTimestamp(parameters.getTimestamp())
                                       .stopMeterValue(Integer.toString(parameters.getMeterStop()))
                                       .stopReason(stopReason)
                                       .eventTimestamp(DateTime.now())
                                       .eventActor(TransactionStopEventActor.station)
                                       .build();

        ocppServerRepository.updateTransaction(params);

        ocppServerRepository.insertMeterValues(chargeBoxIdentity, parameters.getTransactionData(), transactionId);

        applicationEventPublisher.publishEvent(new OcppTransactionEnded(params));
        //------Re-start the new profiles
        startSetProfiles(chargeBoxIdentity);
        //--------------------------------
        return new StopTransactionResponse().withIdTagInfo(idTagInfo);
    }
```

fnkbsi, 2024-01-31:
Hi LydiaBenaida,

Disclaimer: I haven't checked your code in detail. 
But I have an idea that you can check out.
You set the profile before sending the response back, so the charging station may wait for the StartTransaction response.
Are you sending TxProfile or TxDefaultProfile? The TxProfile may be missing the transaction number.

LydiaBenaida, 2024-01-31:
i m sending txProfile for every new transaction