steve-community/steve #1377: how web interface and steve server are communicating with each other

Haleem-khan5, 2024-01-31:
#### Checklist

 I checked other issues already and found no answer
 I checked the documentation and found no answer
 I am running the latest version and the feature

#### Describe the problem you are trying to solve
Actually, I want to start remote transactions from my server (which I have implemented for my application to control the charge point).  To start a remote transaction, I have to access the Steve Web interface. It will send a request to the Steve server, and the server will then send a request to the requested charger.
But I want to send a request from my server to Steve's server for starting a remote transaction and for other functionalities as well.

Where should I find relevant routes for these tasks?
"${ctxPath}/manager/operations/${opVersion}/RemoteStartTransaction"

Any kind of help will be appreciated.
...

juherr, 2024-01-31:
Steve doesn't provide an api for remote start operation.

Check #1291, #1074 and #1000

Haleem-khan5, 2024-02-01:
> Steve doesn't provide an api for remote start operation.
> 
> Check #1291, #1074 and #1000

goekay, 2024-02-01:
hey @Haleem-khan5 as @juherr pointed out, this is currently not implemented. there is a way of doing this. if you know your way around spring framework and RESTful endpoints you could do this.

closed 2024-02-01 by goekay