steve-community/steve #1380: Error migration of schema stevedb to version "0.7.7

maxikys, 2024-02-04:
#### Checklist

- [+ ] I checked other issues already, but found no answer/solution
- [+ ] I checked the documentation and wiki, but found no answer/solution
- [ +] I am running the latest version and the issue still occurs
- [ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.6.0
Operating system  : Ubuntu server 22.04.01
JDK               : 11.1
Database          : mariadb-server-10.6

#### Expected Behavior
Succesfull installation.

#### Actual Behavior

Error during execution of command "sudo ./mvnw package"

#### Steps to Reproduce the Problem

  1. install via Virtual box Ubuntu server
  2. perform adding ssh to virtual box(use port forwarding)
  3. made all instruction like in guid
  

#### Additional context

[WARNING] 
[WARNING] Plugin validation issues were detected in 2 plugin(s)
[WARNING] 
[WARNING]  * pl.project13.maven:git-commit-id-plugin:4.9.10
[WARNING]  * org.flywaydb:flyway-maven-plugin:7.15.0
[WARNING] 
[WARNING] For more or less details, use 'maven.plugin.validation' property with one of the values (case insensitive): [BRIEF, DEFAULT, VERBOSE]
[WARNING] 
[ERROR] Failed to execute goal org.flywaydb:flyway-maven-plugin:7.15.0:migrate (default) on project steve: org.flywaydb.core.internal.command.DbMigrate$FlywayMigrateException: Migration V0_7_7__update.sql failed
[ERROR] -----------------------------------
[ERROR] SQL State  : HY000
[ERROR] Error Code : 1833
[ERROR] Message    : Cannot change column 'chargeBoxId': used in a foreign key constraint 'stevedb/FK_chargeBoxId_c' of table 'stevedb/connector'
[ERROR] Location   : /home/max/steve-steve-3.6.0/src/main/resources/db/migration/V0_7_7__update.sql (/home/max/steve-steve-3.6.0/src/main/resources/db/migration/V0_7_7__update.sql)
[ERROR] Line       : 1
[ERROR] Statement  : ALTER TABLE `chargebox`
[ERROR] CHANGE COLUMN `chargeBoxId` `chargeBoxId` VARCHAR(255) NOT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `endpoint_address` `endpoint_address` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `ocppProtocol` `ocppProtocol` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `chargePointVendor` `chargePointVendor` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `chargePointModel` `chargePointModel` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `chargePointSerialNumber` `chargePointSerialNumber` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `chargeBoxSerialNumber` `chargeBoxSerialNumber` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `fwVersion` `fwVersion` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `fwUpdateStatus` `fwUpdateStatus` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `iccid` `iccid` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `imsi` `imsi` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `meterType` `meterType` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `meterSerialNumber` `meterSerialNumber` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `diagnosticsStatus` `diagnosticsStatus` VARCHAR(255) NULL DEFAULT NULL COMMENT ''
[ERROR] 
[ERROR] -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException

WoCha-FR, 2024-02-05:
I have same problem.
I finally install Steve 3.6.0 on Ubuntu 20.04 LTS following this wiki page : https://github.com/steve-community/steve/wiki/Prepare-Ubuntu-VM-for-SteVe

Schmiedgasse, 2024-02-05:
I have the same problem now.
I used Ubuntu 22.04 and I think the issue is, that mariadb-server is only available in version 10.6

Thanks for your hint, I will try using Ubuntu 20.04

jovrstra, 2024-02-05:
I have the same problem installing Steve using the docker-compose. 
Platform Windows x64, WSL2 with Debian as default distribution.

The container steve-steve-360-app-1 shows the same error messages as in the original post. Not sure if it matters but the last log-entry in mariadb:10.4 container says 'MariaDB init process done. Ready for startup'. Before that is a warning "Aborted connection 8 to db: 'unconnected' user: 'unconnected' host: 'my ip' (This connection closed normally without authentication).

Any solution to this for docker users?
(use different MariaDB version or so?)

Thanks!

maxikys, 2024-02-05:
> I have same problem.
> I finally install Steve 3.6.0 on Ubuntu 20.04 LTS following this wiki page : https://github.com/steve-community/steve/wiki/Prepare-Ubuntu-VM-for-SteVe

maxikys, 2024-02-05:
waiting update

jovrstra, 2024-02-05:
Following the instruction on the Ubuntu LTS installation (https://github.com/steve-community/steve/wiki/Prepare-Ubuntu-VM-for-SteVe ), I changed the MariaDB version to 10.3 (in the docker-compose.yml file). It is still compiling, but judging by the log, it seems to have passed the problem. I will report back.

maxikys, 2024-02-07:
I also successfully install Steve via docker! only change in "docker-compose.yml" version Mariadb to 10.3

closed 2024-02-07 by maxikys

azamau, 2024-02-17:
> Following the instruction on the Ubuntu LTS installation (https://github.com/steve-community/steve/wiki/Prepare-Ubuntu-VM-for-SteVe ), I changed the MariaDB version to 10.3 (in the docker-compose.yml file). It is still compiling, but judging by the log, it seems to have passed the problem. I will report back.
> 
> edit: I can confirm that changing the version of MariaDB in the docker-compose.yml file is sufficient: it compiles completely. The last log entries in the steve container are the different links to connect to the server.

Can you please help me I am running into same problem. I am running Ubuntu 22 and MariaDB is 10.6. the docker-compose.yml file in steve main directory, I changed in line 15 image : 10.3 , but still I am having same issue. 

can you help.

FredBoard, 2024-02-19:
**Same problem on Raspberry pi 5** NVMe SSD:

Linux 6.1.0-rpi8-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux
MySQL versionv10.11.6-MariaDB-0+deb12u1
openjdk version "17.0.10" 2024-01-16
OpenJDK Runtime Environment (build 17.0.10+7-Debian-1deb12u1)
OpenJDK 64-Bit Server VM (build 17.0.10+7-Debian-1deb12u1, mixed mode)
Maven home: /usr/share/maven
Java version: 17.0.10, vendor: Debian, runtime: /usr/lib/jvm/java-17-openjdk-arm64
Default locale: en_GB, platform encoding: UTF-8
OS name: "linux", version: "6.1.0-rpi8-rpi-2712", arch: "aarch64", family: "unix"

:-(

Any patch ?

jovrstra, 2024-02-19:
> Can you please help me I am running into same problem. I am running Ubuntu 22 and MariaDB is 10.6. the docker-compose.yml file in steve main directory, I changed in line 15 image : 10.3 , but still I am having same issue.
> 
> can you help.

For me changing the MariaDB  version on line 15 in the docker-compose.yml file solved the issue, but I'm on Windows with a Debian WSL.  I did start the whole compilation from scratch (so I removed all the containers and volumes that were created, but also images that were pulled).

Still, I guess this issue ought to be reopened...