steve-community/steve #1383: wrong port number in readme

jovrstra, 2024-02-07:
Very small issue:
The readme file on line 137 (First Steps) refers to port 8080 rather than 8180.
:-)

juherr, 2024-02-09:
@jovrstra Why not fix it in a pull request?

jovrstra, 2024-02-09:
Not sure how to do it that way.... I haven't used Github that much. :-)

juherr, 2024-02-09:
https://docs.github.com/en/get-started/start-your-journey/hello-world is a good start.

jovrstra, 2024-02-09:
I'm not planning to develop, I don't need to pull the repository and create a branch... I just needed to install SteVe (in Docker) and wanted to inform about a small thing in the readme.

juherr, 2024-02-10:
In that case, just go to https://github.com/steve-community/steve/blob/master/README.md and click on
[image]

jovrstra, 2024-02-10:
Also requires me to fork it...
Let's just drop it... I just wanted to help.

commit cc4302a2 (2024-02-25): readme nits (closes #1383)
closed 2024-02-25 by goekay