steve-community/steve #1388: DeAuthorized Stop transaction

LydiaBenaida, 2024-02-16:
Hello everyone, 
I'm trying to attribuate a tx profile after the transaction start , but when the charge point takes time to respond to my request , the transaction stopped !!!!

```
    public StartTransactionResponse startTransaction(StartTransactionRequest parameters, String chargeBoxIdentity) {
        if (chargePointUserRepository.isAssociated(chargeBoxIdentity, parameters.getIdTag())) {
            // Get the authorization info of the user, before making tx changes (will affectAuthorizationStatus)
            IdTagInfo info = ocppTagService.getIdTagInfo(
                    parameters.getIdTag(),
                    true,
                    () -> new IdTagInfo().withStatus(AuthorizationStatus.INVALID), null // IdTagInfo is required
            );

            InsertTransactionParams params =
                    InsertTransactionParams.builder()
                            .chargeBoxId(chargeBoxIdentity)
                            .connectorId(parameters.getConnectorId())
                            .idTag(parameters.getIdTag())
                            .startTimestamp(parameters.getTimestamp())
                            .startMeterValue(Integer.toString(parameters.getMeterStart()))
                            .reservationId(parameters.getReservationId())
                            .eventTimestamp(DateTime.now())
                            .build();

            int transactionId = ocppServerRepository.insertTransaction(params);

            applicationEventPublisher.publishEvent(new OcppTransactionStarted(transactionId, params));
            StartTransactionResponse response = new StartTransactionResponse()
                    .withIdTagInfo(info)
                    .withTransactionId(transactionId);

            // Call startSetProfiles after the transaction has started
            startSetProfiles(chargeBoxIdentity,"Start");

            return response;
        }else {
            // If the association is not present, return an "user not allowed" response
            StartTransactionResponse response = new StartTransactionResponse()
                    .withTransactionId(-1) // Use a special value to indicate an error
                    .withIdTagInfo(new IdTagInfo().withStatus(AuthorizationStatus.INVALID));

            return response;
        }

    }
```

lategoodbye, 2024-02-18:
Hi,
there are several reason why you cannot expect a answer here:
- you bypassed the bug report form :-(
- you didn't provide any helpful information
- you reported a problem here which is caused by a charging station
- you attached code without providing any context

goekay, 2024-02-18:
in addition to @lategoodbye 's response, this looks like a duplicate of https://github.com/steve-community/steve/issues/1376 and therefore, i am closing it.

closed 2024-02-18 by goekay