steve-community/steve #1406: charging not started at start time mentioned in the charging profile

Haleem-khan5, 2024-03-11:
I added a charging profile on Steve (you can see the screenshot) and set the charging profile for a specific charger (that is connected to Steve). This set profile was also accepted by the charger. But charging did not start at the mentioned start time in the charging profile (I added this start charging profile time according to utc time ) 

[image]

I did not see any relevant issues here. Can anyone give me guidance on it?

fnkbsi, 2024-03-11:
The "Start Schedule" is the starting point of the profile, not the start of the Charging! (see OCPP 1.6 chapter 7.8 and following) 
To start the charging (current flow) you can try to set the power to zero before the planned start of the charging. (A Transaction must be active). The success of such a approach depends on the EVSE and the EV. 

Another note: the first “start period” of a schedule should be zero.

Haleem-khan5, 2024-03-11:
Why does the transaction have to be active? When I am trying to set the charging profile for future charger scheduling,
Actually, i want to set the charging profile and using this charger profile, i want to automatically start the charging
How can I achieve this?

fnkbsi, 2024-03-11:
the transaction must be active when the charging shall start (authentication, authorization). The tx_default_profiles can be set without an active transaction.