steve-community/steve #1417: Can't compile SteVe 3.6.0

d6e6t6a6l6e6s, 2024-03-23:
**Hello, I installed a new server and I can't compile SteVe 3.6.0. What happened ? Looks like I do the same thing...**

SteVe Version     : SteVe 3.6.0
Operating system  : Debian 11 bullseye
JDK  : openjdk version "11.0.22" 2024-01-16
Database : 10.5.23-MariaDB-0+deb11u1 Debian 11

```
#  ./mvnw package
Warning: JAVA_HOME environment variable is not set.
[INFO] Scanning for projects...
[INFO]
[INFO] -------------------------< de.rwth.idsg:steve >-------------------------
[INFO] Building SteVe 3.6.0
[INFO]   from pom.xml
[INFO] --------------------------------[ jar ]---------------------------------
[INFO]
[INFO] --- enforcer:3.3.0:enforce (enforce-java) @ steve ---
[INFO] Rule 0: org.apache.maven.enforcer.rules.version.RequireJavaVersion passed
[INFO]
[INFO] --- git-commit-id:4.9.10:revision (default) @ steve ---
[INFO]
[INFO] --- properties:1.1.0:read-project-properties (default) @ steve ---
[INFO]
[INFO] --- flyway:7.15.0:migrate (default) @ steve ---
[INFO] Flyway Community Edition 7.15.0 by Redgate
[INFO] Database: jdbc:mysql://localhost:3306/stevedb (MySQL 5.5)
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  01:10 min
[INFO] Finished at: 2024-03-23T10:10:57Z
[INFO] ------------------------------------------------------------------------
[WARNING]
[WARNING] Plugin validation issues were detected in 2 plugin(s)
[WARNING]
[WARNING]  * pl.project13.maven:git-commit-id-plugin:4.9.10
[WARNING]  * org.flywaydb:flyway-maven-plugin:7.15.0
[WARNING]
[WARNING] For more or less details, use 'maven.plugin.validation' property with one of the values (case insensitive): [BRIEF, DEFAULT, VERBOSE]
[WARNING]
[ERROR] Failed to execute goal org.flywaydb:flyway-maven-plugin:7.15.0:migrate (default) on project steve: org.flywaydb.core.api.exception.Flyw                                      ayValidateException: Validate failed: Migrations have failed validation
[ERROR] Detected failed migration to version 0.7.7 (update). Please remove any half-completed changes then run repair to fix the schema history                                      .
[ERROR] Need more flexibility with validation rules? Learn more: https://rd.gt/3AbJUZE
[ERROR] -> [Help 1]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException
```

fnkbsi, 2024-03-25:
Maybe you find some hints at issue #1380.

d6e6t6a6l6e6s, 2024-03-29:
I tried but no luck :(

fnkbsi, 2024-03-30:
have you seen the pk #1394?

d6e6t6a6l6e6s, 2024-04-01:
Now I can only successfully compile _Steve 3.6.0_ with **Ubuntu 20.04.2 LTS** and **MariaDB 10.3.39**. I think the problem is that only MariaDB version 10.3 is supported. However, I can't figure out how I created a Virtual Machine with Ubuntu 22.04.3 LTS and MariaDB 10.6.12 six months ago. This virtual machine is still running. However, there is no way I can create a new one.
[image]

juherr, 2024-04-02:
@d6e6t6a6l6e6s Steve should work with java 17 (and 21) + ubuntu-22.04 + mariadb:10.6.14. Check according actions:  https://github.com/steve-community/steve/actions/runs/8498526126/job/23278326051

closed 2025-09-15 by goekay