steve-community/steve #1424: Charger not following profiles

jovrstra, 2024-03-27:
Hello,
I have a weird problem; I don't think it is Steve related but I hope someone can shed some light... 
We have an old prototype chademo V2G charger (so for sure not on the supported chargers list, but it is ocpp 1.6 compliant) and it seems to work. However, the charger does not follow the profiles that are defined for it. We see that it accepts a profile when it is sent to the charger (both in Steve and in the log of the charger), but when starting a transaction it always falls back to its default charging (10kW). It is not a time issue (we defined profiles to identify a potential shift in hours, due to UTC), and it occurs for all three types of profiles, relative and absolute. It is also is independent of starting the transaction on the charger with an RFID (authenticated via Steve) or performing a remote start on Steve (both approaches yield a transaction that is visible in Steve).
Perhaps one indicator... The charger should have a pulsating led when it is smart charging, and this does not happen: it is solidly lit. I'm currently using a first generation Leaf to test it. Once we had a newer Leaf and I thought that with that car it was pulsating (but I may remember wrong and I don't have that car right now to test). Can it be car dependent?
Thanks for any suggestions.
Jörg

goekay, 2025-08-29:
hey there, this is obviously a "charger problem" and not a "steve problem". still, i left the issue open for folks to chime in which did not happen unfortunately. i am closing this issue due to inactivity.

jovrstra, 2025-09-01:
Hello,
I forgot I posted the question here, but can confirm it was a pure "charger problem". While compatible with some aspectes of OCPP 1.6, the charger uses a proprietary way of receiving profiles.
So 100% confirmed it is not a "steve problem". :-)
Thanks for leaving it open for people to see it and possible chime in.
Jörg