steve-community/steve #1431: Is it possible to only allow charging for added OCPP tags?

surreal123, 2024-04-05:
Hi there,

sorry if this may not be the right place to ask this question.
We run Steve version 3.5.0. 

Is it possible (by configuration) to prohibit charging for OCPP tags which weren't added?
At the moment every OCPP Tag can register itself an start charging.

I did not find anything about this. I have seen configuration keys such as:
FreeCharging, FreeChargingMode and RfidTagFreeCharging but can't find further info.

As far as i know setting FreeCharging to false (in the charge points) would result in prohibiting charging even for already added/known OCPP tags.

Best regards

Gunob17, 2024-04-11:
Hi.
No setting FreeChargning to false should not result in all tags beeing rejected as long as it can validate the tag with a backend operator, through there might depend on what chargepoint box you are using as there might be a setting around explicitly authenticating with ocpp that needs to be enabled, this could be the case if the chargepoint also contains a local whitelist.  
Its been a while since I played around with SteVe but as far as i remember it will not return "accepted" to a chargepoint if the ocpp tag is not in its "Ocpp tag" list and if it is not there it should be added to the "unknown tags" list. maybe check if your chargebox is connected in the log and sending the tag information.

goekay, 2024-05-20:
> as far as i remember it will not return "accepted" to a chargepoint if the ocpp tag is not in its "Ocpp tag" list 

this is true. therefore, i cannot understand the following:

> At the moment every OCPP Tag can register itself an start charging.

your situation is probably highly dependent on your station's behaviour and configuration. i would suggest to get to know your station and investigate its behaviour/configuration.

closed 2024-05-20 by goekay