steve-community/steve #1463: Not able to install on Ubuntu with Mariadb 10.11

VidarSandvik, 2024-05-26:
Are any organisations using the Steve OCPP server in operation?
Is it any developer still actively working to solve any issues comming?

#### Checklist
- [X] I checked other issues already, but found no answer/solution
- [X] I checked the documentation and wiki, but found no answer/solution
- [X] I am running the latest version and the issue still occurs
- [X] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications

SteVe Version     : 3.6.0 (Latest)
Operating system  : Ubuntu 22.04
JDK               : OpenJDK 17
Database          : MariDB 10.11

goekay, 2024-05-26:
> Are any organisations using the Steve OCPP server in operation?

yes

> Is it any developer still actively working to solve any issues comming?

yes, including me. most of the issues are in two categories: 

1. user problem: the user does not know what he/she is doing, and is not executing some steps properly due to lack of experience etc. 
2. not a steve issue: the problem is not in this application but rather with the network, configuration, and more often than not due to the station not doing the proper thing.

> Not able to install on Ubuntu with Mariadb 10.11

when we do not even announce that steve supports mariadb 10.11, why do you try to use it and complain when it does not work? refer to [this](https://github.com/steve-community/steve/blob/master/README.md#system-requirements) and mariadb versions that are currently supported.

VidarSandvik, 2024-05-26:
Thanks for your reply. We are considering what path to go for our OCP cloud development and are considering the Steve OCPP server sourcse. We were just getting concerned when we saw the number of open posts and also that recent stable of by example MariaDB is supported.

goekay, 2024-05-26:
> * [x]  I checked the documentation and wiki, but found no answer/solution
> * [x]  I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

you even checked these two boxes which i use as filter for me to decide _how much of my attention i should give you_: if you check a box and therefore claim `X` but your message and issue report exhibit that you are in the category of `NOT X`, then i make a conscious decision: 

A. i either ignore your message (mostly the case recently), or 
B. reply with an aggressive message signaling you are an idiot, or 
C. help you out showing the right way but suffer personally because i had to do customer service for an idiot.

the tone of the original message is a really strong guide for me to decide a path. i might think you are an idiot, but if you are nice i will choose C. this path requires a lot of mental energy from me which i dont always have though.

today and with this issue, i think you deserve B. but i am not being nice, if i choose the path B. therefore, i try to avoid it. because of these reasons i don't go with B and C, leaving only A as option. that is the reason for having this many open issues.

> Is it any developer still actively working to solve any issues comming?

there you go. i solved the issue: read the readme and understand what you are doing before opening an issue. and do not check these boxes blindly.

i hope this was helpful.

closed 2024-05-26 by goekay