steve-community/steve #1464: Kafka Integration

mks1991, 2024-05-28:
I want to integrate with Kafka for performing all OCPP command and received meter values.
#### Checklist

- [ ] I checked other issues already and found no answer
- [ ] I checked the documentation and found no answer
- [ ] I am running the latest version and the feature i am requesting is not implemented there

#### Describe the problem you are trying to solve

...

#### Describe the solution you'd like

...

#### Describe alternative solutions or features you've considered 

...

#### Additional context

...

goekay, 2024-05-28:
sure, please go ahead then.

what is your issue or request with us?

ManojRailbit, 2024-06-27:
I want manage web sockets and perform all OCPP commands using kafka.

juherr, 2024-06-27:
Steve does not function as an OCPP gateway. 
I recommend referring to the GitHub repository at https://github.com/aws-samples/aws-ocpp-gateway and adapting your code to use Kafka instead of MQTT.

closed 2024-08-07 by goekay