steve-community/steve #1465: Approval request: steve logo variation and placement

gruselhaus, 2024-05-30:
Hi guys,

We would like to use SteVe for a large deployment with EV stations at our headquarters and would be happy if we could refer to you. We would like to add a SteVe plot to the Mennekes station but can only plot an all black version. Can you approve the logo version and placement?

Thank you very much for your work,

Nico

[image]

(Original image: https://www.roenne-technik.de/media/de/marke_hersteller_produkte/mennekes/neuheiten/amedio-professional/csm_mennekes_work_dsf7129_72e03854aa.jpg)

goekay, 2024-06-02:
hey there, thank for getting in touch for that kind of a request, and wanting to add the logo to the chargers! this never really happens (really).

i can approve the colors and the placement of the logo. 

and again, thanks.

closed 2024-06-02 by gruselhaus

goekay, 2024-06-02:
pls post photos of chargers (if you are allowed) with logos when they are deployed :) this is a rare occurrence.

gruselhaus, 2024-06-03:
will do