steve-community/steve #1475: Charger connections limit

kassymbekoff, 2024-06-04:
Hello everyone!
Is there any limit of websocket connections by default in SteVe? Or how can I increase it?
Problem: when connected more than 100 chargers 4-5 chargers randomly always go offline

tefelberson, 2024-06-14:
I would also like to connect several charging points to Steve. How did you do ? what tools do you use?

goekay, 2024-09-20:
> Problem: when connected more than 100 chargers 4-5 chargers randomly always go offline

there is no limit we set in code/config. you can have as many charging stations as you like in your database, and you can have as many open websocket connections as the memory resources allow: each websocket connection will eat from JVM memory, and at some point (_X_ open connections), you might hit the ceiling.

however, if you are not running with some _reaaaaally_ conservative memory allocation to steve, this ceiling should not be around 100 chargers. therefore, can you elaborate? logs? exception stack traces?

goekay, 2025-03-17:
closing due to inactivity.

closed 2025-03-17 by goekay