steve-community/steve #1477: Date handling in OCPP operations

jerome-benoit, 2024-06-06:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [x] I am running the latest version and the issue still occurs
- [x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.6.0
```

#### Description 

The date input in several forms for OCPP operations is not translated to the UTC in the payload. The OCPP server time stamping is already correctly using UTC.
The charging station expects date in UTC after translation of the input date from the CSMS timezone to the charging station timezone. The timezone information is not expected to be part of it.

'SetChargingProfile` or `UpdateFirmware` operations are impacted. The later is not possible since the input check for date in the future if you're running the CSMS on a GMT+X timezone.

goekay, 2024-06-07:
> if you're running the CSMS on a GMT+X timezone.

so, you are running steve not in the default (i.e. UTC) but some other timezone?

jerome-benoit, 2024-06-07:
> > if you're running the CSMS on a GMT+X timezone.
> 
> so, you are running steve not in the default (i.e. UTC) but some other timezone?

The date inputs are always assuming the OS timezone while the server expects UTC dates. It can't work properly if the date inputs are not translated to UTC, which is not the case currently. 
Either the date inputs are documented to input UTC date (few work), either they are translated to UTC given the CSMS OS timezone and charging station timezone (a lot of work).

jerome-benoit, 2024-06-11:
For now, I've commented out all `@Future` date validation in order to be able to input UTC dates in UTC+ timezone. 

If it's considered as reasonable solution for now, I can make a PR.