steve-community/steve #1482: Network requirement question

gruselhaus, 2024-06-11:
I have a specific question about network requirements, but first let me outline our deployment strategy. 

We are in the process of rolling out 30+ charge points in the field with no wired network uplink from the transformer stations to the nearest network takeover point on the property.

So the choice is mobile LTE routers that tunnel directly into the data center where SteVe is running. However, we do not have a site-to-site VPN available, so although the charging stations reach SteVe, requests from SteVe are not routed directly to the charging station network.

Hence my question: do the charging stations have to be accessible from the network where SteVe is running or do the charging stations establish the connection?

Sorry if the question is relatively easy to answer, OCPP is quite new to us.

Nico

juherr, 2024-06-12:
Hi,

It depends on the station and the ocpp version it supports.

With soap transport (ocpp before 1.6), the station must be reachable.
With json/websocket transport (ocpp 1.6 and later), the station initiates the 2-side connection.
But some manufacturers may have specific features outside ocpp that need the station to be directly reachable from the network.

gruselhaus, 2024-06-12:
According to the Mennekes documentation, the Amedia Professional+ should be compatible with OCPP 1.6 and 1.6j.  @juherr  thank's for your quick response. Will report back...

closed 2024-06-13 by gruselhaus