steve-community/steve #1490: Connect charging station to Steve.

tefelberson, 2024-06-16:
Sure! Here is the translated and corrected text in English:

Hello everyone,

I am starting my research project on the cybersecurity of electric vehicle charging stations. I have taken the time to study and understand how the OCPP protocol works, and I need a simulation lab. **Steve** is one of the most popular (if not the most popular) open-source CSMS. I would like to connect charging stations to Steve. I have tried **Everest** and **MicroOCPP**, but I have not yet been able to successfully communicate with Steve using these tools. Can you help me? Could you provide me with the steps to follow?

EasonChen11, 2024-07-27:
Hello,

I noticed your previous inquiry regarding the setup issues you experienced while trying to connect charging stations to SteVe using Everest and MicroOCPP. I am currently facing similar challenges. Could you please share if you were able to resolve the issue? If so, would you mind providing some guidance or the steps you followed to achieve successful communication? Any help would be greatly appreciated as it could greatly assist with my project.

Thank you!

mohammdkhalilii, 2024-08-01:
Same, I have bugs on connecting via WebSocket, I was trying on my local network, and it works totally fine.
But when I go to public and build SteVe on public IP server, it did not work unfortunately.
I am now trying to debug it and know that where is the problem.