steve-community/steve #1491: Multiple Webusers login

nimraidris, 2024-06-17:
I added a new user's username and password using the database and enabled the authority, but I am unable to log in with this user. I can only log in with the auth.user and auth.password mentioned in the main properties. However, I can see my new web user in Data Management > Web Users.

fnkbsi, 2024-06-19:
Are you revering to the pull request #1165 (web user  sound like that) or steve master? 
For the PR:  Use SteVe (additional sites are part of the PR) to add the web user, because only a hash of the password will saved.

nimraidris, 2024-06-19:
I created a new user through Steve still it's not working I already followed the steps given in #1165.

[image]

[image]

fnkbsi, 2024-06-21:
It works on my system, so I don't  know what's happening on your system.

The DB migration script should have create an webuser 'admin' with the password 'pass'. Did this webuser exist and could you enter the websites with those credentials? 

Your first webuser seems to have a short password entry. Can you enter the website with those credentials?

nimraidris, 2024-06-24:
@fnkbsi Yeah, it works for me I deleted the target and reinstall again, and I used #1165 as mentioned.

goekay, 2024-08-07:
closing the issue, since it looks like resolved.

closed 2024-08-07 by goekay