steve-community/steve #1502: When i am trying to run the steve.jar file, i am getting below mentioned error:

rizkytri17-eng, 2024-07-02:
C:\Users\engin\steve>java -jar target/steve.jar
Log file: C:\Users\engin\logs\steve.log
Starting. FAILED!
Please refer to the log file for details
Exception in thread "main" java.util.concurrent.ExecutionException: java.lang.IllegalStateException: STOPPED
        at java.base/java.util.concurrent.CompletableFuture.reportGet(CompletableFuture.java:396)
        at java.base/java.util.concurrent.CompletableFuture.get(CompletableFuture.java:2096)
        at org.eclipse.jetty.server.Server.doStop(Server.java:490)
        at org.eclipse.jetty.util.component.AbstractLifeCycle.stop(AbstractLifeCycle.java:132)
        at de.rwth.idsg.steve.JettyServer.stop(JettyServer.java:185)
        at de.rwth.idsg.steve.SteveProdStarter.stop(SteveProdStarter.java:80)
        at de.rwth.idsg.steve.Application.stop(Application.java:85)
        at de.rwth.idsg.steve.Application.main(Application.java:68)
Caused by: java.lang.IllegalStateException: STOPPED
        at org.eclipse.jetty.server.handler.ContextHandler.shutdown(ContextHandler.java:774)
        at java.base/java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:197)
        at java.base/java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1708)
        at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:509)
        at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:499)
        at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:575)
        at java.base/java.util.stream.AbstractPipeline.evaluateToArrayNode(AbstractPipeline.java:260)
        at java.base/java.util.stream.ReferencePipeline.toArray(ReferencePipeline.java:616)
        at org.eclipse.jetty.util.component.Graceful.shutdown(Graceful.java:146)
        ... 6 more
[image]

rizkytri17-eng, 2024-07-02:
I attached steve.log
[steve.log](https://github.com/user-attachments/files/16061283/steve.log)

goekay, 2024-07-14:
it is not possible to understand the reason for steve stopping. the logs etc are not helpful unfortunately.

pls check your application.properties. maybe IP or port are occupied by another app etc.

closed 2024-09-20 by goekay