steve-community/steve #1503: Logs are Unavailable

lmccarty5, 2024-07-03:
SteVe Version     : 3.6.0
Operating system  : Ubuntu 22.04 Virtual Machine Running on Windows 11 Host
JDK               :  java 21.0.2  on host

#### Expected Behavior

when hitting the log button on the steve web interface, a log will appear

#### Actual Behavior

a mostly blank page appears that says steve log unavailable
[image]

#### Steps to Reproduce the Problem

  1. open steve directory
  2. start docker 
  3. open web interface
  4. hit logs button in top right
  5. get logs unavailable error

juherr, 2024-07-04:
When running Steve with Docker, there is currently no log available in the web interface. 

This is in line with the expected behavior when running "cloud" services, as outlined in the 12-factor methodology for building software-as-a-service applications. 
To address this, it is recommended to redirect the image output into a dedicated log service. 

Furthermore, it may be advisable to hide the log link when it is not available.

lmccarty5, 2024-07-05:
Do you have any documentation or resources that could help me with redirecting the image output to a dedicated log service?

juherr, 2024-07-05:
You'll find many articles.

One of them https://sharmarajdaksh.github.io/blog/shoving-your-docker-logs-to-elk

commit 10d58f57 (2024-07-14): better msg if logs are unavailable (closes #1503)
commit e60eaf31 (2024-07-14): Merge pull request #1511 from steve-community/1503-logs-are-unavailable

better msg if logs are unavailable (closes #1503)
closed 2024-07-14 by goekay